bulletphysics/bullet3 #1980: Tracking errors with inverse dynamics

huaijiangzhu, 2018-11-05:
Hi,

I have been trying position tracking with inverse dynamics on the kuka iiwa. I noticed that the tracking error increases towards the end of the chain: joint 0 has reasonable tracking performance while joint 6 is completely off.

[image]

The controller I have been using is a simple inner-outer-loop inverse dynamics controller as presented in Robot Dynamics and Control by Mark W. Spong (2nd edition). The same controller gives nearly perfect tracking when computing the resulted acceleration with ABA and forward integrating the position via another rigid body dynamics library (Pinocchio: https://stack-of-tasks.github.io/pinocchio/)

[image]

Did anyone come across similar problems?

In the attachment, you can find the jupyter notebook I used to create the plot.
Test environment:
Ubuntu 16.04.05
Python 3.5.2
Pybullet 2.1.9 installed via pip.
[inverse_dynamics_kuka.tar.gz](https://github.com/bulletphysics/bullet3/files/2549682/inverse_dynamics_kuka.tar.gz)

huaijiangzhu, 2018-11-07:
This comes from the effect of joint damping. Setting the joint damping to zero or adding an additional damping compensation term to the controller solves the problem.

closed 2018-11-07 by huaijiangzhu

erwincoumans, 2018-11-07:
Thanks for the update. Did you use changeDynamics(bodyUid, linkIndex, linearDamping=0, angularDamping=0)?

huaijiangzhu, 2018-11-07:
> Thanks for the update. Did you use changeDynamics(bodyUid, linkIndex, linearDamping=0, angularDamping=0)?

No. That function would change the link damping while I need to change the joint damping. There seems to be no direct way to do that in pybullet so I just changed the URDF.

MichaelLutter, 2019-01-05:
If I understand this correctly, the forward dynamics simulation uses the static and damping friction of the URDF, while the inverse dynamics computation ignores the friction.

Can you add this to Pybullet Quickstart Guide in the InverseDynamics section, because this information is quite important for robot control?

erwincoumans, 2019-12-08:
Somehow I didn't get to address this closed issue, but it is now: changeDynamics has the option to set jointDamping now.
Also, I added the following snippet in the [PyBullet Quickstart Guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#).
~~~~~~~~~~~~~~~~
lso note that calculateInverseDynamics ignores the joint/link damping, while forward dynamics (in stepSimulation) includes those damping terms. So if you want to compare the inverse dynamics and forward dynamics, make sure to set those damping terms to zero using changeDynamics with jointDamping and link damping through linearDamping and angularDamping.
~~~~~~~~~~~~~~~~