bulletphysics/bullet3 #1991: Simulating Flexible Joints

MichaelLutter, 2018-11-14:
Can one simulate flexible joints using Pybullet? More concretely can one simulate two joints that are coupled by a massless spring? Therefore, the torque for the coupled joint would be modelled as 

\tau_{motor} = J_{motor} qdd_{motor} + K (q_{joint} - q_{motor})
\tau_{joint} = K (q_{joint} - q_{motor})

where \tau is the motor / joint torque and q, qd, qdd are the joint / motor positions, velocities and accelerations. One could simulate the motor outside of PyBullet, but this risks that the integration of qdd_{motor} and qdd_{joint} would diverge and yield "unrealistic" simulations.

erwincoumans, 2018-12-22:
No, we do this motor modeling outside of PyBullet.

erwincoumans, 2019-02-26:
Now I realize you can simulate flexible joints using PyBullet.createUserConstraint.

You can create a 'fixed' constraint with a limited force, to mimic a flexible joint.

ac-93, 2020-04-22:
> If the maximum torque is not enough to model the spring, we can expose the stiffness and damping (which gets converted into CFM and ERP internally) to model a the spring force.

I'd like to adjust the stiffness and damping coefficients for a project I'm working on, would it be possible to expose these using something like the p.changeConstraint()? or is this already exposed through the erp parameter?

Thanks

closed 2021-04-26 by erwincoumans