bulletphysics/bullet3 #1992: pkg-config does not account for libtbb

stolk, 2018-11-15:
If you build bullet with multithreading support using TBB, like so:

```
cmake \
        -DBULLET2_MULTITHREADING=ON \
        -DBULLET2_USE_TBB_MULTITHREADING=ON \
        -DCMAKE_BUILD_TYPE=Release ../
```

Then if you link your app using `pkg-config --libs bullet` then libtbb is not used.

```
$ pkg-config --libs bullet
-L/usr/local/lib -lBulletSoftBody -lBulletDynamics -lBulletCollision -lLinearMath
```

It's easy to fix by adding -ltbb to your link command line, but if pkg-config knew about libtbb it would be better.

erwincoumans, 2018-11-15:
We don't enable multithreading/tbb by default, and I don't want to make it a hard dependency.
Is there anything we can do to allow for tbb, but not fail if the user doesn't use it/have it not installed?

stolk, 2018-11-15:
Of course it should only show up as a dep for the app when bullet it built against it.

CMake already correctly searches for libtbb, if you ask for TBB, and will report failure if the lib is not there.

It's a really minor thing, but the point is: if bullet uses tbb, it would be best if pkg-config knew about this.

That way, you can simply link your app with `pkg-config --libs bullet` and regardless how bullet was configured, it will always work if you use pkg-config.

But again: really minor thing. I'll take a quick look at how the pkg-config is generated, and if it is easy to dynamically augment the deps, depending on CMake's flags.

scpeters, 2018-11-15:
@stolk does https://github.com/scpeters/bullet3/commit/cf90a52b370d7944edec050d9f79fa533f112a30 work? I'll submit a pull request if so

erwincoumans, 2018-11-15:
Thanks Steven. Is there also a way to add the preprocessor definitions in pkg-config?
(I never used pkg-config, it was a contribution)

scpeters, 2018-11-15:
yeah, we just need to add those definitions to a cmake variable and then put it between `@` characters in the `.pc` template

stolk, 2018-11-15:
tbb deps works for me:
```
$ pkg-config --libs bullet
-L/usr/local/lib -lBulletSoftBody -lBulletDynamics -lBulletCollision -lLinearMath -ltbb -latomic
```
Thanks!

erwincoumans, 2018-12-22:
If someone has a patch for pkg-config, please consider sharing a pull request. 
Until then, let's close the issue (there are no volunteers working on it).

closed 2018-12-22 by erwincoumans

scpeters, 2019-01-04:
The double-precision flag is already set in the pc file with `@BULLET_DOUBLE_DEF@`, which is set in the CMakeLists.txt:

* https://github.com/bulletphysics/bullet3/blob/233a7083d19ded151e80c2d9c7cd3d28856d6f76/CMakeLists.txt#L217

It would be a simple matter to set the other flags in cmake variable and add them surrounded by `@` in the `.pc` template