bulletphysics/bullet3 #2003: Crash In Running App_BulletExampleBrowser_vs2010_debug.exe

yangxiaopo, 2018-11-23:
In File bullet3\examples\OpenGLWindow\GLRenderToTexture.cpp
void GLRenderToTexture::init(int width, int height, GLuint textureId, int renderTextureType)
glFramebufferTexture(GL_FRAMEBUFFER, GL_DEPTH_ATTACHMENT, textureId, 0);

[image]
[image]
thank you, please ;)

yangxiaopo, 2018-11-23:
may be my opengl error?

yangxiaopo, 2018-11-23:
[image]

erwincoumans, 2018-11-23:
Upgrade your GPU drivers, or add a proper discrete GPU in your machine.

closed 2018-11-23 by erwincoumans

yangxiaopo, 2018-11-23:
ok