bulletphysics/bullet3 #2004: Parallel convexSweepTests

akortunov, 2018-11-26:
Currently in OpenMW we use the Bullet for collision detection.
Basically, we have a btCollisionWorld with a lot of static compound shapes (based on triangles) and couple of dozens of dynamic moving actors (which use capsules or box shapes).
To detect collisions, we iterate over actors and use the convexSweepTest for every actor in the single thread.
With many actors in area or with not fast CPU performance can be pretty low.

Is there a correct way to batch convexSweepTests?
I found [this](https://www.reddit.com/r/gamedev/comments/7nv541/bullet_physics_and_multiple_threads/) thread on Reddit, where user built Bullet with the DBULLET2_USE_THREAD_LOCKS=ON and just executed convexSweepTest in different threads, but I do not know if it is a proper solution.

BluePrintRandom, 2018-11-26:
Using a sensor sphere test VS simplified bounds should return the shapes
the capsule could be colliding with

Only checking this data would remove much of your itteration

In upbge we recently setup bvhtree so we can use it to get overlaps quickly.
(so simplified bounds check yields potential collision bodies, and bvhtree
returns overlapping leaves)

<https://www.reddit.com/r/gamedev/comments/7nv541/bullet_physics_and_multiple_threads/>

psi29a, 2018-11-26:
@BluePrintRandom You mind explaining things a bit more, many of the terms you are using are flying over my head and like that of others?

Also, the question that was asked hasn't yet be answered. :)

closed 2019-10-27 by akortunov