bulletphysics/bullet3 #2006: Cannot load URDF file

yumion, 2018-11-27:
Hello, this is my first issue. 
I tried using own URDF(like car) and reinforcement learning on pybullet.
This task is that the car reach a target. 
When training episode is 69, I got this error `Cannot load URDF file.` every time.
```
---------------------------------------------------------------------------
error                                     Traceback (most recent call last)
<ipython-input-7-49574b19e137> in <module>
      8     exploration_final_eps=0.02,
      9     print_freq=1,
---> 10     callback=callback
     11 )
     12 print("Saving model to test_car_model.pkl")

~/.pyenv/versions/anaconda3-5.3.0/lib/python3.6/site-packages/baselines/deepq/simple.py in learn(env, q_func, lr, max_timesteps, buffer_size, exploration_fraction, exploration_final_eps, train_freq, batch_size, print_freq, checkpoint_freq, learning_starts, gamma, target_network_update_freq, prioritized_replay, prioritized_replay_alpha, prioritized_replay_beta0, prioritized_replay_beta_iters, prioritized_replay_eps, param_noise, callback)
    249             episode_rewards[-1] += rew
    250             if done:
--> 251                 obs = env.reset()
    252                 episode_rewards.append(0.0)
    253                 reset = True

<ipython-input-2-7799453c3644> in reset(self)
     31         #フィールドを表示
     32         p.setGravity(0,0,-10)
---> 33         self.planeId = p.loadURDF("plane100.urdf")
     34 
     35         #オブジェクトモデルを表示

error: Cannot load URDF file.
```
This is my code.
```
import pybullet as p
import time
import numpy as np
import gym
from gym import spaces
...

class Test_car(gym.Env):
    
    def __init__(self):
        print("init")
        super().__init__()
        self.episodes = 0
        self.max_steps = 10
        self.height = 64
        self.width = 64
        self.action_space = spaces.Discrete(4) #前後左右
        self.observation_space = spaces.Box(0, 255, [self.height, self.width, 3]) #Boxは連続値
        self.reward_range = [-1,1]
        '''pybullet側の初期設定'''
        p.connect(p.DIRECT)
        p.setAdditionalSearchPath("../ros_ws/src/test_car_description/urdf/")
        self.maxForce = 10
        self.reset()
        print("init_reset終了")
            
    def reset(self):
        print("====episode:"+str(self.episodes)+"=================")
        self.episodes += 1
        self.steps = 0
        targetX, targetY = np.random.permutation(np.arange(10))[0:2]
        self.targetPos = [targetX, targetY, 0]
        
        '''pybullet側'''
        #bulletの世界をリセット
        p.resetSimulation()
        #フィールドを表示
        p.setGravity(0,0,-10)
        self.planeId = p.loadURDF("plane100.urdf")
        
        #オブジェクトモデルを表示
        self.startPos = [0,0,0]
        self.startOrientation = p.getQuaternionFromEuler([0,0,0])
        self.car = p.loadURDF("test_car.urdf", self.startPos, self.startOrientation)
        
        # ターゲットを表示
        self.target = p.createCollisionShape(
            p.GEOM_CYLINDER, radius=0.2, height=2, collisionFramePosition=self.targetPos)
        p.createMultiBody(0, self.target)
        
        return self.observation()
        
    def step(self, action):
        print("---step:"+str(self.steps)+"-------")
        self.steps += 1
        if action == 0:
            ...
        for i in range(200):
            p.stepSimulation()
            time.sleep(1./240.)
        
        observation = self.observation()
        done = self.is_done()
        reward = self.reward()
        return observation, reward, done, {}
      
    def render(self, mode='rgb_array', close=False):
        if mode != "rgb_array":
            return np.array([])
        base_pos, orn = p.getBasePositionAndOrientation(self.car)
        cam_eye = np.array(base_pos) + [0.1,0,0.2]
        cam_target = np.array(base_pos) + [2,0,0.2]
        cam_upvec = [1,0,1]

        view_matrix = p.computeViewMatrix(
                cameraEyePosition=cam_eye,
                cameraTargetPosition=cam_target,
                cameraUpVector=cam_upvec)

        proj_matrix = p.computeProjectionMatrixFOV(
            fov=60, aspect=float(self.width)/self.height,
            nearVal=0.1, farVal=100.0)

        (_, _, rgb, _, mask) = p.getCameraImage(
            width=self.width, height=self.height, viewMatrix=view_matrix,
            projectionMatrix=proj_matrix, renderer=p.ER_BULLET_HARDWARE_OPENGL)

        rgb_array = np.array(rgb)
        rgb_array = rgb_array[:,:,:3]
        mask_array = np.array(mask)

        return rgb_array
...
    def is_done(self):
        frame = self.render()
        self.area = self.calc_area(frame)
        if self.area >= 80:
            done = True
        elif self.steps > self.max_steps:
            done = True
        else:
            done = False
        return done

    def reward(self):
        if self.area >= 80:
            reward = 1
        elif self.steps > self.max_steps:
            reward = -1
        else:
            reward = 0
        print("reward: ", reward)
        return reward

env = Test_car()

from baselines import deepq
...
model = deepq.models.cnn_to_mlp([(512,5,1)], [256,64,4])
act = deepq.learn(
    env,
    q_func=model,
    lr=1e-2,
    max_timesteps=10000,
    buffer_size=50000,
    exploration_fraction=0.1,
    exploration_final_eps=0.02,
    print_freq=1,
    callback=callback
)
```
I can't understand where is wrong code, so I wonder if this is a bug on pybullet.
Please tell me the way to solve this problem.

Sincerely.

erwincoumans, 2018-11-27:
Something could be wrong with the URDF. Is there any warning in the console while loading the URDF?
You could only load the URDF and saveState, and then use restoreState instead?
Please share a full project including all assets (URDF, obj/stl, textures etc) if you are still stuck.

yumion, 2018-11-28:
I use the URDF file named "plane100.urdf" included pybullet_data.
> Is there any warning in the console while loading the URDF?
No. There is only that error.

>You could only load the URDF and saveState, and then use restoreState instead?
I want to load a target URDF at random after saveState, so I think it's difficult to use saveState and restoreState.

Here is my project. Please give me advises.
[my_project.zip](https://github.com/bulletphysics/bullet3/files/2623357/my_project.zip)

yumion, 2018-11-30:
I solved this problem.
My own URDF file's **mesh file** is wrong. 
I usually use "123D Design" to make STL.  If I change the application to "Fusion 360", there is no error in bullet.

Thank you your advise!

closed 2018-11-30 by yumion