bulletphysics/bullet3 #2009: batch create collision objects faster.

pgruenbacher, 2018-11-28:
I'm coming from the godot world. Create 3,000 collision objects at once leads to > 5 seconds of delay. What bothers me is that the initial added objects take < 20-40 microsecs to create. However that addition time increases linearly to around 700 microsecs to create. I'm able to confirm by commenting the 
`btCollisionWorld::addCollisionObject()` that the processing time is increasing within bullet itself. I couldn't find anywhere in the godot code that indicates it's querying the state between time I create a new object... I cannot find anywhere in docs about batch creating collision objects (of same filter, group, aabb collision shape).

erwincoumans, 2018-11-28:
Do you create them all at the same location? The Bullet Example Browser has some example with 1000 objects, it shouldn't take that long. Recreate the issue in the ExampleBrowser, then we can check it.

BluePrintRandom, 2018-11-28:
batch add object would be useful in upbge as well, I need it for terrain
tiles loading in all their content as the tile loads in.

your sure the time is not coming from rebuilding the scenegraph?

pgruenbacher, 2018-11-28:
I did create them all in the same location. cause i was doing create 3000 objects, then iterate over them to transform. I updated the code to add a random transofrm when creating the 3000 objects, and now the delay still increases linearly, but slower this time (200 microsec at the end instead of 900 microsec).

pgruenbacher, 2018-11-28:
hm I'm unable to replicate with the 3000 box demo. I changed the dynamic rigid bodies to static and changed func to addCollisionObject but the demo still loads fast...

pgruenbacher, 2018-11-28:
er yes, by random i mean I shift the transform position by +3,+3 eahc time i add, and the shape only has a 0.4 radius so they shouldn't be overlapping.

But what I don't understand, is bullet actually evaluating collisions / overlap each time I create a new object and add to the world? why wouldn't that just happen in the next physics step?

On a side-tangent @erwincoumans , let's say I have 3000 agents and I just want to test ray-traces and areas on the AABB, but have no need for the precision of evaulating a collision shape. I currently do a collision box shape, is there a way for bullet3 to do just aabb collision objects without a shape so as to achieve a potential speedup?

pgruenbacher, 2018-11-28:
yes it would seem the world is configured to use btDbvBroadphase. 
```
	dispatcher = bulletnew(GodotCollisionDispatcher(collisionConfiguration));
	broadphase = bulletnew(btDbvtBroadphase);
	solver = bulletnew(btSequentialImpulseConstraintSolver);

	if (p_create_soft_world) {
		dynamicsWorld = new (world_mem) btSoftRigidDynamicsWorld(dispatcher, broadphase, solver, collisionConfiguration);
		soft_body_world_info = bulletnew(btSoftBodyWorldInfo);
	} else {
		dynamicsWorld = new (world_mem) btDiscreteDynamicsWorld(dispatcher, broadphase, solver, collisionConfiguration);
	}```

erwincoumans, 2018-11-28:
>> hm I'm unable to replicate with the 3000 box demo

As soon as you can create the issue in BulletExampleBrowser we can look at it. Otherwise it is likely related to code inside Godot.

pgruenbacher, 2018-11-28:
let's assume closed for now.

bdancer, 2020-03-22:
@pgruenbacherI'm struggling with the same issue (also not sure about your exact case...) and I've figured out that setting `btDbvtBroadphase::m_deferedcollide=true` makes things much faster.
Also doing `world->getCollisionObjectArray().clear();` before `delete world` makes freeing world instant.