bulletphysics/bullet3 #2011: PyBullet GUI mode through SSH results in Segmentation Fault

Olimoyo, 2018-11-29:
Hi,

I am trying to run PyBullet in GUI mode on an external computer which I'm SSHed to, so that I can render and see the GUI on my laptop. I am able to run glxgears succesfully so I think X forwarding is working fine. 

When I try to connect (in GUI mode) with PyBullet I get the following error:
```
pybullet build time: Nov 29 2018 14:42:05
current_dir=/home/olimoyo/.local/lib/python3.5/site-packages/pybullet_envs/bullet
options= --opengl2
argv[0]=--opengl2
startThreads creating 1 threads.
starting thread 0
started thread 0
argc=3
argv[0] = --unused
argv[1] = --opengl2
argv[2] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
4

	visual 0x17e selected
GL_VENDOR=ATI Technologies Inc.
GL_RENDERER=AMD Radeon Pro 450 OpenGL Engine
GL_VERSION=1.4 (2.1 ATI-2.2.8)
GL_SHADING_LANGUAGE_VERSION=1.20
pthread_getconcurrency()=0
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0
MotionThreadFunc thread started
[Obelisk:20700] *** Process received signal ***
[Obelisk:20700] Signal: Segmentation fault (11)
[Obelisk:20700] Signal code: Address not mapped (1)
[Obelisk:20700] Failing at address: (nil)
[Obelisk:20700] [ 0] /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390)[0x7f8990e29390]
[Obelisk:20700] *** End of error message ***
Segmentation fault (core dumped)
```

Does anyone have an idea what the problem is?
Has anyone being able to succesfully forward the PyBullet GUI through SSH?

erwincoumans, 2018-11-29:
We use PyBullet in GUI mode over Remote Chrome Desktop with a GPU driver that supports OpenGL 3.2+. It looks like your driver only supports GL_VERSION=1.4 , which is not sufficient.

--opengl2 may not be fully supported by your driver. Try getting a full callstack (debug build), or add printf to check out where the crash happens.

You can run headless and use the simpler CPU renderer (TinyRenderer) and dump images, for debugging?

I also ran the PyBullet client remotely and connect to a local PyBullet server, and connect over TCP or UDP over an ssh tunnel, but then your assets needs to be available locally as well (URDF, obj etc).

Otherwise, just compile and run PyBullet locally on your laptop.

Olimoyo, 2018-11-30:
Thanks for the quick response!

You were right, I tested with another client laptop and this time it did work, although it was super slow (I guess that's just the limitations of SSH).

The TCP or UDP solution sounds interesting. I'm not too familiar with this stuff, do you happen to have an example which I can refer to?

My local laptop isn't as powerful as the server so I wanted to offload all the computing to it and to only visualize the GUI on my laptop. It's also a shared lab computer so I wanted to avoid using it remotely.

Yup, it's for running reinforcement learning experiments. Dumping images could be an alternative, but I was kind of hoping for a solution which allowed a "live stream" of the GUI.

erwincoumans, 2018-12-04:
This is a GPU driver issue, we cannot fix it. Let's close the issue.

closed 2018-12-04 by erwincoumans