bulletphysics/bullet3 #2014: PyBullet GUI freeze in MacOS 10.14.1

yijiangh, 2018-11-30:
OS: MacOS Mojave 10.14.1
Python version: 2.7.10
Pybullet version: 2.4.0

Installing pybullet (2.4.0) succeeded by `pip install pybullet`.

However, when I tried to test pybullet's visualization GUI by
```python
import pybullet
pybullet.connect(pybullet.GUI)
```
I got:
```
Version = 4.1 INTEL-12.2.17
Vendor = Intel Inc.
Renderer = Intel(R) UHD Graphics 617
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
0
```

And the invoked example browser is white and frozen. I have to force-quit to get rid of the frozen window:

[image]

Have anyone else experienced the same problem? Is this issue potentially related to the fact that Apple [stops supporting OpenGL app](https://stackoverflow.com/questions/52509427/mac-mojave-opengl) starting from 2018?

Thanks!

erwincoumans, 2018-12-01:
This is expected. You need to call 'stepSimulation' or other PyBullet command to refresh drawing.
Problem on Mac OSX is that OpenGL commands HAVE to be send from the main thread. Python is the main thread. See the problem?

closed 2018-12-01 by erwincoumans

yijiangh, 2018-12-04:
Thanks for your reply! The weird thing is that when I tried to run the pybullet example files, it still gives me that white screen...

[image]

However, after I resize the window (accidentally), the example browser seems to be refreshed and everything works like a charm.

I'm not sure what's reason for this but just wanna drop some notes here in case someone encounters the similar question in the future.

Thanks.

erwincoumans, 2019-01-28:
It works just fine with me (just upgraded some machines to Mojave. Perhaps they fixed things in some updates?

yijiangh, 2019-02-03:
@erwincoumans Yes, just tested -  Mojave 10.14.2 fixed this! Thanks!

theophanemayaud, 2021-11-30:
Is it just me or has this problem come back with Monterey ? I seem to get a frozen (spinner) pybullet whenever python is not calling simulation updates... 
However when calling pybullet.stepSimulation() often, the simulation, and zoom/moving around the view are possible.

jasonjabbour, 2021-12-21:
@theophanemayaud I'm not exactly sure what Monterey is but the PyBullet GUI is frozen for me when I try to use the addUserDebugparameter(). My code was working a few weeks ago with no issues. I made no changes, and the GUI is now frozen. I'm not sure what happened. Did you find a fix to your problem?

theophanemayaud, 2021-12-21:
It’s the latest version of macOS ! No, I didn’t find any 🤷‍♂️

Flacedoo, 2022-04-14:
I actually just experienced the exact same behaviour when switching to another MacBook, this time a 2019 pro 16inch. Newest macOS installed (12.3.1), freshly installed pybullet. When launching the gui, the window freezes, as soon as I resize it it works without any problem. But I narrowed it down a bit on my side: this only happens when my display link dock is connected? As soon as I unplug it everything works directly. When plugged in, the window freezes. When moving the window to one of the external screens, even resizing doesn’t work and the window is frozen at all times. Moving it back to the internal screen makes it work again? 

Hint: I also have the exact same setup with another MacBook working, without any problems. Could this also be related to the graphics chip from AMD which is only installed in the MacBook with problems?

jasonjabbour, 2022-04-14:
@Flacedoo Glad to hear someone else faced the same issue. Thanks for narrowing it down. When I faced this issue, I always had my laptop connected to an external display. It's very possible that the external display contributed to the freezing issue. For anyone else who might face this issue in the future, I was also using an AMD processor and graphics chip. 

My laptop with an Intel processor does not have this issue!

cemoke, 2022-10-10:
> 
Have you found a solution? 
I am using Monterey with M1, and I have the same problem. Unless I call stepSimulation() nothing gets updated.