bulletphysics/bullet3 #2020: [pybullet] setCollisionFilterGroupMask doesn't work on loaded object with multiple links

teristam, 2018-12-06:
Hi, 

In the `collisionFilter.py`, the collision filter works on simple objects. However, when I load a more complex object, the `setCollisionFilterGroupMask` doesn't seem to work anymore. The links are always set to collide no matter how I set the filter group and mask. Is there anything I have done wrong here?

```
import pybullet as p
import time
import pybullet_data

p.connect(p.GUI)
p.setAdditionalSearchPath(pybullet_data.getDataPath())

planeId = p.loadURDF("plane.urdf", useMaximalCoordinates=False)
robot = p.loadURDF('biped/biped2d_pybullet.urdf', [0,0,3], useMaximalCoordinates=False)
cubeId = p.loadURDF("cube_collisionfilter.urdf", [1,0,3], useMaximalCoordinates=False)

collisionFilterGroup = 0x2
collisionFilterMask = 0x2
for i in range(p.getNumJoints(robot)):
    p.setCollisionFilterGroupMask(robot,i,collisionFilterGroup,collisionFilterMask)

for i in range(p.getNumJoints(cubeId)):
    p.setCollisionFilterGroupMask(cubeId,i,collisionFilterGroup,collisionFilterMask)

collisionFilterGroup2 = 0x1
collisionFilterMask2 = 0x1 
p.setCollisionFilterGroupMask(planeId,-1,collisionFilterGroup2,collisionFilterMask2)

p.setRealTimeSimulation(1)
p.setGravity(0,0,-10)
while (p.isConnected()):
	time.sleep(1./240.)
	p.setGravity(0,0,-10)
```

erwincoumans, 2019-02-19:
Fixed in this CL:
https://github.com/bulletphysics/bullet3/compare/master...erwincoumans:master

Thanks for the report!

closed 2019-02-19 by erwincoumans

FengJi2021, 2022-01-05:
i still want to report this issue again as i tried with my own model and set them in groups with different masks but still collides while i tried with cube it doesnt.