bulletphysics/bullet3 #2023: App_robotSimulator crash with "btSpinMutex::lock(): Assertion `!"unimplemented btSpinMutex::lock() called"' failed in debug mode."

eastskykang, 2018-12-07:
Hello, 

As I run App_RobotSimulator with cmake debug mode, I got the following error. 

```
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=Intel Open Source Technology Center
GL_RENDERER=Mesa DRI Intel(R) HD Graphics 630 (Kaby Lake GT2) 
GL_VERSION=4.5 (Core Profile) Mesa 18.0.5
GL_SHADING_LANGUAGE_VERSION=4.50
pthread_getconcurrency()=0
Version = 4.5 (Core Profile) Mesa 18.0.5
Vendor = Intel Open Source Technology Center
Renderer = Mesa DRI Intel(R) HD Graphics 630 (Kaby Lake GT2) 
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
ven = Intel Open Source Technology Center
Workaround for some crash in the Intel OpenGL driver on Linux/Ubuntu
App_RobotSimulator_d: /home/donghok/git/bullet3/src/LinearMath/btThreads.cpp:198: void btSpinMutex::lock(): Assertion `!"unimplemented btSpinMutex::lock() called"' failed.

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
```

Someone can give some suggestion to fix it? 

Thank you.

erwincoumans, 2018-12-08:
How did you build the exe? cmake or premake? If one, can you try the other?

AlCap23, 2018-12-13:
I have a similar error while loading a urdf using the b3RobotSimulatorClientAPI_noGUI .

I pretty much copied the code from the [example library](https://github.com/bulletphysics/bullet3/blob/master/examples/RobotSimulator/RobotSimulatorMain.cpp).

```c
    model = new b3RobotSimulatorClientAPI_NoGUI();
    connected = model->connect(eCONNECT_DIRECT);
    if (!connected)
    {
        std::cerr << "Error: Could not connect to bullet server..." << std::endl;
        abort();
    }

    // Config for bullet
    //model->configureDebugVisualizer(COV_ENABLE_GUI, 0);
    model->setTimeOut(10);
    model->syncBodies();

    // Set the time step
    btScalar bStepSize = 1. / 240.;
    model->setTimeStep(bStepSize);
    // Convert the gravity
    btVector3 bGravity = btVector3(0, 0, 0);
    //for (int i = 0; i < 3; i++)
    //{
    //    bGravity[i] = modelConfig["gravity"][i].as<double>();
    //}
    model->setGravity(bGravity);
    std::cout << "Working until load URDF" << std::endl;
    // load the urdf
    model->loadURDF(modelFile);
    for (int i = 0; i < 100; i++)
    {
        model->stepSimulation();
    }
    std::cout << "Working after load URDF" << std::endl;
```

Which leads to the following output
```
Working until load URDF
dynamics_benchmark:BULLET_INSTALL/bullet/src/LinearMath/btThreads.cpp:198: void btSpinMutex::lock(): Assertion `!"unimplemented btSpinMutex::lock() called"' failed.
Aborted (core dumped)
```

Running gdb gives:
```
Working until load URDF
dynamics_benchmark: /media/jmartensen/Data/general/research/01_benchmark/external/bullet/src/LinearMath/btThreads.cpp:198: void btSpinMutex::lock(): Assertion `!"unimplemented btSpinMutex::lock() called"' failed.

Program received signal SIGABRT, Aborted.
0x00007ffff5c89428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) quit()
A debugging session is active.
```

I am using cmake ( C++11) , Ubuntu 16.04.

erwincoumans, 2018-12-22:
Can you use premake to build the project and make sure to enable_multithreading?
~~~~~~~~~~~~~~~~~
premake4  --double --enable_multithreading  --targetdir="../bin" vs2010 
~~~~~~~~~~~~~~~~~

When using cmake, please use BULLET2_MULTITHREADING=ON

It is a harmless warning if using Bullet in a single-threaded environment (DIRECT mode /NO_GUI).
I'll look into fixing this.

closed 2018-12-22 by erwincoumans