bulletphysics/bullet3 #2024: terminate called after throwing an instance of 'std::bad_alloc'

abefetterman, 2018-12-11:
I am running into a crash when running the latest version of pybullet for a long time (500k renders / 5m timesteps). With 2.4.0 from pip:

`[ps5mmr7tw:22530] *** Process received signal ***
[ps5mmr7tw:22530] Signal: Segmentation fault (11)
[ps5mmr7tw:22530] Signal code: Address not mapped (1)
[ps5mmr7tw:22530] Failing at address: 0x55e5f7a2d5b0
[ps5mmr7tw:22530] [ 0] /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390)[0x7f9cfaea6390]
[ps5mmr7tw:22530] [ 1] /home/abe/anaconda3/lib/python3.6/site-packages/pybullet.cpython-36m-x86_64-linux-gnu.so(_ZN6Shader8fragmentE3vecILm3EfER8TGAColor+0x2f0)[0x7f9cb6a071a0]
[ps5mmr7tw:22530] [ 2] /home/abe/anaconda3/lib/python3.6/site-packages/pybullet.cpython-36m-x86_64-linux-gnu.so(_Z15triangleClippedR3matILm4ELm3EfES1_R7IShaderR8TGAImagePfPiRKS_ILm4ELm4EfEi+0x71a)[0x7f9cb69fedaa]
[ps5mmr7tw:22530] [ 3] /home/abe/anaconda3/lib/python3.6/site-packages/pybullet.cpython-36m-x86_64-linux-gnu.so(_ZN12TinyRenderer12renderObjectER20TinyRenderObjectData+0x3073)[0x7f9cb6a045a3]
[ps5mmr7tw:22530] [ 4] /home/abe/anaconda3/lib/python3.6/site-packages/pybullet.cpython-36m-x86_64-linux-gnu.so(_ZN32TinyRendererVisualShapeConverter6renderEPKfS1_+0xab0)[0x7f9cb6a9c9d0]
[ps5mmr7tw:22530] [ 5] /home/abe/anaconda3/lib/python3.6/site-packages/pybullet.cpython-36m-x86_64-linux-gnu.so(_ZN29PhysicsServerCommandProcessor32processRequestCameraImageCommandERK19SharedMemoryCommandR18SharedMemoryStatusPci+0x367)[0x7f9cb6a3a1a7]
[ps5mmr7tw:22530] [ 6] /home/abe/anaconda3/lib/python3.6/site-packages/pybullet.cpython-36m-x86_64-linux-gnu.so(_ZN29PhysicsServerCommandProcessor14processCommandERK19SharedMemoryCommandR18SharedMemoryStatusPci+0x4f6)[0x7f9cb6a60aa6]
[ps5mmr7tw:22530] [ 7] /home/abe/anaconda3/lib/python3.6/site-packages/pybullet.cpython-36m-x86_64-linux-gnu.so(_ZN13PhysicsDirect13processCameraERK19SharedMemoryCommand+0x86)[0x7f9cb6a34d26]
[ps5mmr7tw:22530] [ 8] /home/abe/anaconda3/lib/python3.6/site-packages/pybullet.cpython-36m-x86_64-linux-gnu.so(b3SubmitClientCommand+0x14)[0x7f9cb6a8b9c4]
[ps5mmr7tw:22530] [ 9] /home/abe/anaconda3/lib/python3.6/site-packages/pybullet.cpython-36m-x86_64-linux-gnu.so(b3SubmitClientCommandAndWaitStatus+0x80)[0x7f9cb6a8ba60]
[ps5mmr7tw:22530] [10] /home/abe/anaconda3/lib/python3.6/site-packages/pybullet.cpython-36m-x86_64-linux-gnu.so(+0x1a3c0b)[0x7f9cb69e9c0b]
[ps5mmr7tw:22530] [11] python3(_PyCFunction_FastCallDict+0x154)[0x55e7f1909364]
[ps5mmr7tw:22530] [12] python3(_PyObject_FastCallDict+0x2bf)[0x55e7f190977f]
[ps5mmr7tw:22530] [13] python3(+0x2053c2)[0x55e7f1a023c2]
[ps5mmr7tw:22530] [14] python3(_PyObject_FastCallDict+0x8b)[0x55e7f190954b]
[ps5mmr7tw:22530] [15] python3(_PyObject_FastCallKeywords+0xaa)[0x55e7f19960aa]
[ps5mmr7tw:22530] [16] python3(+0x19f00e)[0x55e7f199c00e]
[ps5mmr7tw:22530] [17] python3(_PyEval_EvalFrameDefault+0x10c7)[0x55e7f19be3e7]
[ps5mmr7tw:22530] [18] python3(+0x198b8b)[0x55e7f1995b8b]
[ps5mmr7tw:22530] [19] python3(+0x19ef95)[0x55e7f199bf95]
[ps5mmr7tw:22530] [20] python3(_PyEval_EvalFrameDefault+0x30a)[0x55e7f19bd62a]
[ps5mmr7tw:22530] [21] python3(+0x198b8b)[0x55e7f1995b8b]
[ps5mmr7tw:22530] [22] python3(+0x19ef95)[0x55e7f199bf95]
[ps5mmr7tw:22530] [23] python3(_PyEval_EvalFrameDefault+0x30a)[0x55e7f19bd62a]
[ps5mmr7tw:22530] [24] python3(+0x198b8b)[0x55e7f1995b8b]
[ps5mmr7tw:22530] [25] python3(+0x19ef95)[0x55e7f199bf95]
[ps5mmr7tw:22530] [26] python3(_PyEval_EvalFrameDefault+0x30a)[0x55e7f19bd62a]
[ps5mmr7tw:22530] [27] python3(+0x198b8b)[0x55e7f1995b8b]
[ps5mmr7tw:22530] [28] python3(+0x19ef95)[0x55e7f199bf95]
[ps5mmr7tw:22530] [29] python3(_PyEval_EvalFrameDefault+0x30a)[0x55e7f19bd62a]
[ps5mmr7tw:22530] *** End of error message ***`

With pybullet.so built from source:

`terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[ps5mmr7tw:03449] *** Process received signal ***
[ps5mmr7tw:03449] Signal: Aborted (6)
[ps5mmr7tw:03449] Signal code:  (-6)
[ps5mmr7tw:03449] [ 0] /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390)[0x7f58e3d7b390]
[ps5mmr7tw:03449] [ 1] /lib/x86_64-linux-gnu/libc.so.6(gsignal+0x38)[0x7f58e39d5428]
[ps5mmr7tw:03449] [ 2] /lib/x86_64-linux-gnu/libc.so.6(abort+0x16a)[0x7f58e39d702a]
[ps5mmr7tw:03449] [ 3] /home/abe/anaconda3/lib/python3.6/site-packages/scipy/sparse/../../../../libstdc++.so.6(_ZN9__gnu_cxx27__verbose_terminate_handlerEv+0xbc)[0x7f58d1efcb39]
[ps5mmr7tw:03449] [ 4] /home/abe/anaconda3/lib/python3.6/site-packages/scipy/sparse/../../../../libstdc++.so.6(+0x941fb)[0x7f58d1efb1fb]
[ps5mmr7tw:03449] [ 5] /home/abe/anaconda3/lib/python3.6/site-packages/scipy/sparse/../../../../libstdc++.so.6(+0x94234)[0x7f58d1efb234]
[ps5mmr7tw:03449] [ 6] /home/abe/anaconda3/lib/python3.6/site-packages/scipy/sparse/../../../../libstdc++.so.6(+0xafc8a)[0x7f58d1f16c8a]
[ps5mmr7tw:03449] [ 7] /lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7f58e3d716ba]
[ps5mmr7tw:03449] [ 8] /lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7f58e3aa741d]
[ps5mmr7tw:03449] *** End of error message ***`

Versions: Ubuntu 16.04, libc6 2.23-0ubuntu10

I do not get these errors with pybullet 2.3.9 from pip.

Any ideas for what is going on?

erwincoumans, 2018-12-12:
It is not clear what is happening. Can you try out 2.4.1 (uploaded to https://pypi.org/project/pybullet/)

Otherwise, please share a small complete reproduction case.

abefetterman, 2018-12-13:
2.4.1 didn't solve the issue, but switching to pybullet.ER_TINY_RENDERER did. I'll close this until I can create a small complete reproduction case.

closed 2018-12-13 by abefetterman