bulletphysics/bullet3 #2031: Weighted Axis can easily result in null vector and cause assert error

adnanmunawar, 2018-12-21:
https://github.com/bulletphysics/bullet3/blob/4a66d6c80b38a87ecbdf2fd5c4d281f0b5913d22/src/BulletDynamics/ConstraintSolver/btHingeConstraint.cpp#L766

If the two bodies have equal masses and axis pointing in the opposite directions, ax1 = (0,0,0) in which cause it won't normalize on the successive line of code and break the joint and succesive joints.

erwincoumans, 2018-12-21:
Good point. We could check for near degenerate ax1, and if the length is close to zero, use factA=0, factB=1.
Do you mind testing that and create a patch?

adnanmunawar, 2018-12-22:
Thanks for pointing out the possible solutions, I have worked around it but I shall go ahead and test the patch you suggested and then submit it as a PR

erwincoumans, 2018-12-22:
Here is some suggested fix:
https://github.com/bulletphysics/bullet3/pull/2030/commits/63ac2182ae7fecfb552b3837aa631e4d84ddc210
Please re-open if that doesn't address it.

closed 2018-12-22 by erwincoumans