bulletphysics/bullet3 #2032: Heightfield and raycast performance

Zylann, 2018-12-21:
I came across this issue in Godot engine, which uses Bullet for its heightmap shape: https://github.com/godotengine/godot/issues/24390

Details are in the ticket linked above, but in short: I found raycasting a heightfield is tremendously slow. The reason is, it seems rays are enclosed in a AABB which is then used to query every single triangle lying inside it. Tell me if I'm wrong, but this is highly suboptimal for a collider of this kind, to the point even running a supernaive method like [this one](https://github.com/Zylann/godot_heightmap_plugin/blob/master/addons/zylann.hterrain/hterrain.gd#L1056) outperforms it, and not even using C++ (i.e advancing a point by small amounts until it is under the terrain).

I'm not be the first one who bumped into this, so I saw various workarounds, such as chunking the heightfield into many smaller ones, or chunking the ray in many smaller ones. But both these solutions are impractical, it's something we *systematically* would have to do, and kind of defeats the point in having a heightfield shape in the first place.

Another method would be to represent the heightfield with many chunked AABBs enclosing chunks, rather than a single huge one, without touching the underlying data grid, however that would have to happen internally.

Is there any plan to improve this?

erwincoumans, 2018-12-21:
Thanks for the report. It should be easy to add some acceleration structure indeed, for example using the btDbvt dynamic aabb tree. Perhaps there is already a pull request or patch somewhere.

>> Is there any plan to improve this?

This project relies purely on volunteers adding/fixing things and contributing patches. If you have time, consider adding the feature.

erwincoumans, 2018-12-23:
If you volunteer improving it, a pull request if welcome.
Here is some old discussion too:
https://github.com/bulletphysics/bullet3/issues/73
Thanks!

Zylann, 2018-12-23:
Cool, thanks for digging it up :) I'll have a look when I get some time.

Zylann, 2018-12-30:
It looks like the old PR got archived, and files I can download are not patches, just the whole files as they were years ago... quite hard to find out what actually happened

Zylann, 2018-12-30:
I think I got the PR to work within the Bullet version in Godot Engine (not so far from the latest).
It uses a quad tree accelerator, but I think there are a few ways to improve it:
- It currently has an early-exit for short rays, but it tests 3D length. I would test 2D length instead so long vertical rays should not travel the tree (which is nice for vehicles or typical is-grounded checks).
- When reaching a leaf node, it still ends up checking all triangles within a quad. I would instead walk through the projected ray with a kind of bresenham algorithm, so we can make the tree less deep while not impacting the final test too much.
- I don't know if it works on non-square, or uneven terrains, so for now I trust the PR until I test this case

I may port it to this repository, but for now you can find it here: https://github.com/Zylann/godot/commit/f07458555091a5450d9c6c5acb8c53b8896c5134

Zylann, 2019-01-01:
I experimented another technique (from the points above), which speeds up raycasts a lot even without data structure (so even if there is no accelerator), which can be found here https://github.com/Zylann/godot/commit/e780160d99d83a34a31f66470f5f4d832fcff9db

I need to learn how to test this from within the Bullet repo so I can do a PR here, is there a testbed I can use for testing heightfields?

lunkhound, 2019-01-02:
I don't think any of the demos that come with Bullet use the heightfield shape.
However, there is a demo called the benchmark demo that tests ray casting against a terrain. Code for it is [here](https://github.com/bulletphysics/bullet3/blob/master/examples/Benchmarks/BenchmarkDemo.cpp).

In that demo, the terrain is a triangle mesh. You could modify that to (optionally) use a heightfield in place of the triangle mesh.

erwincoumans, 2019-01-03:
We haven't ported the heightfield shape to the BulletExampleBrowser yet.

Here is the old code, from the glut based demos:
https://github.com/bulletphysics/bullet3/blob/old_demos/Demos/TerrainDemo/TerrainDemo.cpp

closed 2019-04-16 by erwincoumans