bulletphysics/bullet3 #2033: High mass objects bubble into groundplane.

stolk, 2018-12-21:
If you use high mass values, the solver will no longer keep objects separated from a static ground plane.

In video: left side has density 1.0 shapes. The right hand side shows same simulation, but with density 80.0 shapes.

https://youtu.be/zaa-X3cyEtI

To reproduce: add a scale of 80.0f in line 1319 of BenchmarkDemo.cpp

```
bram@Desktop64:~/src/bullet3$ git diff examples/Benchmarks/BenchmarkDemo.cpp
diff --git a/examples/Benchmarks/BenchmarkDemo.cpp b/examples/Benchmarks/BenchmarkDemo.cpp
index 110a35eab..9798565e1 100644
--- a/examples/Benchmarks/BenchmarkDemo.cpp
+++ b/examples/Benchmarks/BenchmarkDemo.cpp
@@ -1316,7 +1316,7 @@ void BenchmarkDemo::createTest8()
                btTransform transform;
                transform.setIdentity();
                transform.setOrigin(btVector3(origin[0],origin[1]+fallHeight,origin[2]));
-               const float mass = halton_volu[i];
+               const float mass = 80.0f * halton_volu[i];
                btVector3 inertia(0,0,0);
                shp->calculateLocalInertia(mass, inertia);
                btRigidBody::btRigidBodyConstructionInfo ci(mass, 0, shp, inertia);

```

erwincoumans, 2018-12-21:
The globalCFM is causing this. Comment out this line:
~~~~~~~~~~~~~~
//si.m_globalCfm = 0.015f;
~~~~~~~~~~~~~~

stolk, 2018-12-21:
Thanks Erwin,

Quite so.

In my game, I also override the btManifoldPoint's ERP and CFM, but I guess those are separate from the global value.

I Added a little Cfm to make it more stable, so I'll probably keep the globalCfm and lower the density of the shapes.

closed 2018-12-21 by stolk