bulletphysics/bullet3 #2035: A strange problem

pyni, 2018-12-28:
Hi, i am working on a grasping problem with pybullet.
However, we the object is grasped by the gripper, the object will rotate itself and fall down. This is very strange:
[image]
 [image]
[image]
[image]
[image]

Why???
I have set the friction. Any suggestions please?

erwincoumans, 2019-01-03:
Some tips: try using box collision shapes (in case you are using meshes/convex hulls)
use a much higher number of solver iterations
use a smaller time step
What are the friction coefficients for gripper and box? lateral friction, rolling friction and spinning friction?
What are the masses of the gripper and box?
How is the gripper actuated? position control, velocity control or force/torque control?
Try using smaller forces (position and velocity control has a maximum force field)

pyni, 2019-01-03:
OK，thanks a lot.
I will try it.

closed 2019-01-15 by erwincoumans