bulletphysics/bullet3 #2040: Symbol not found for architecture x86_64 brew install

msaada, 2019-01-03:
I everyone, I'm trying to use bullet in my project but I cannot build it because some functions seems to be missing.

```
Undefined symbols for architecture x86_64:
  "btRigidBody::setMassProps(double, btVector3 const&)", referenced from:
      SetupSolver::maintain(int) in SetupSolver.cpp.o
      SetupSolver::unmaintain(int) in SetupSolver.cpp.o
      SetupSolver::strip(Tooth*, double, double) in SetupSolver.cpp.o
  "btConvexHullShape::btConvexHullShape(double const*, int, int)", referenced from:
      AlignerSolver::initScene() in AlignerSolver.cpp.o
      SetupSolver::addMesh(QVtkManipulator&, Tooth*, bool) in SetupSolver.cpp.o
      SetupSolver::strip(Tooth*, double, double) in SetupSolver.cpp.o
  "btStaticPlaneShape::btStaticPlaneShape(btVector3 const&, double)", referenced from:
      SetupSolver::addStaticPlane(btVector3 const&, double const&) in SetupSolver.cpp.o
ld: symbol(s) not found for architecture x86_64
```
I'm on MacOS 10.14.2. I installed bullet with brew and my cmake version is 3.13.2

Does anyone know how can I resolve this issue ?

Many thanks 
Mickaël

erwincoumans, 2019-01-03:
Do you have a small reproduction case mini project with cmake? Without that we cannot help.

erwincoumans, 2019-01-04:
Please re-open if you provide a repro case that breaks with latest github version.

closed 2019-01-04 by erwincoumans