bulletphysics/bullet3 #2048: Wiki is not working on pybullet.org

MrOnlineCoder, 2019-01-05:
All links to Bullet wiki are broken - for example, going to https://pybullet.org/mediawiki-1.5.8/index.php/Glossary_of_Terms gives 404 error:

> Not Found
The requested URL /mediawiki-1.5.8/index.php/Glossary_of_Terms was not found on this server.
Apache/2.4.18 (Ubuntu) Server at pybullet.org Port 443

MrOnlineCoder, 2019-01-09:
We can also move/copy some articles from wiki to the GitHub wiki, so it will be accessible all the time.

erwincoumans, 2019-01-15:
Indeed, it was too much work to convert the old wiki to new server. Will get to it eventually.

closed 2019-01-15 by erwincoumans

etc0de, 2020-03-23:
Maybe you could set up an empty wiki, and encourage people to manually move it over or something? It would be useful to have it back in some shape or form

Scylardor, 2022-05-03:
> 

I second that.
If at some point, the maintainers of this repo are willing to open a new Wiki (why not on this Github repo?), I will be interested to join the effort of migrating the old wiki to the new one.