bulletphysics/bullet3 #2057: How to do simulation of flexible object in

houliwei, 2019-01-14:


houliwei, 2019-01-14:
Can pybullet model a flexible object with certain stiffness ？

erwincoumans, 2019-01-31:
Some people used the deformable simulation in PyBullet, see https://sites.google.com/view/sim-to-real-deformable
We don't have many resources for this, so we focus on the rigid/multibody/collision detection parts.

erwincoumans, 2019-02-26:
You can also use createConstraint with a fixed constraint using limited force to mimic soft constraints.
See the constraint.zip example here: https://github.com/bulletphysics/bullet3/issues/2100#issuecomment-461621661

DanielTakeshi, 2021-03-01:
@houliwei  We have been able to use deformables in PyBullet in this recent work: https://berkeleyautomation.github.io/bags/

closed 2021-04-26 by erwincoumans