bulletphysics/bullet3 #2064: Collision margins for object insertion

arpastrana, 2019-01-17:
I need to perform the robotic assembly of two beams in **pybullet**: one complete beam should be inserted into the other's notch. I have simplified the task to just one beam falling into the other to investigate what the insertion threshold would be (please see picture below).  

The beams' cross-section is **0.04 x 0.12 m**. They are loaded from URDF files via STL meshes. The STLs are directly exported in meters from trimesh.

[image]

## Expected Result
Giving the notch a **0.04** width to achieve a _flushed_ fit (equal to the other beam's width), the falling beam should fit into the other one with some frictional resistance.

## Actual Result
Flush-notched beam doesn't go in.

I looked in the forum and found that there is a default _collision margin_in **pybullet** . I tested four different notch widths in **0.002 m** intervals (**0.040, 0.042, 0.044, 0.046 m**) and let the beam fall down. Only the last two widths turned into a successful insertion. I have also modified the lateral friction from default **0.5 to 0.1** to see if that had an effect on the insertion, but it was unnoticeable.  Time step was set to 1/24 seconds. 

The files for the reproduction case are here: [beam_collision.zip](https://github.com/bulletphysics/bullet3/files/2769286/beam_collision.zip)

## Questions
- What is the collision margin in the **pybullet**? 
- Is there a method to adjust this value? I looked in the documentation but seems that this is not exposed in the Python version.
- Do the STL units have an impact on the collision detection in **pybullet**? Most of the examples I saw in this repo apply a downsizing scale factor in the URDF definition.

Ultimately, I seek to have control over different notches widths to measure the variations in force-torque sensor reading in one of a robot's joints, using a simulation time step of 1/240 s.

erwincoumans, 2019-01-24:
The default collision margin in PyBullet is 0.001 meter, which is 0.1 mm. 
Since the margin is for both objects, and happens on both sides, 4 times the collision margin (0.004) sounds about right. You need to give some room for contact to happen.

1/24 -> this is way too big, please use the default timestep of 240 Hz or smaller.

closed 2019-01-24 by erwincoumans

arpastrana, 2019-01-24:
Hi @erwincoumans, many thanks for your the clarification.

So the collision margin is 0.001 meter = 1.0 millimeter (hope the 0.1 mm was a typo). Is this units' independent? For instance, if I scaled my model 1000 times, the collision margin would be 0.001 kilometers= 1.0 meter.

Lastly and more importantly for me, is it possible for the user to explicitly change the collision margin in PyBullet via `p.setPhysicsEngineParameter` or similar? Say, not to go smaller than the default value of  0.001 (I guess this is a threshold value for minimum numerical stability), but to increase it?

erwincoumans, 2019-01-24:
collision margin is 0.001 units, or 1mm indeed, assuming SI units, meter, seconds. There is currently no API to change it, such API can be added.

v-wewei, 2020-01-16:
@erwincoumans could you pls add a python API for adjust the collision margin for us ?

lorepieri8, 2021-02-17:
@erwincoumans Are there any plans for adding collision margin in the python API? Thanks

Guptajakala, 2021-04-26:
+1
Peg in hole experiments can hardly be conducted right now.

jackson-liugang, 2022-05-30:
can peg in hole experiments be conducted in pybullet?