bulletphysics/bullet3 #2068: btVector3 use of union with __m128 causes suboptimal code gen

TrianglesPCT, 2019-01-19:
Here is a godbolt link demonstrating: https://godbolt.org/z/gyyvcz

You can set USE_UNION to 0 to see how it improves code gen

Fixing this is a fairly simple change for SSE, in btVector3 just need to replace all uses of m_floats with mVec128.m128_f32, the compiler is smart enough to issue the optimal instructions(_mm_cvtss_f32 for x, vshufps for the others).

 It would probably would be best to replace many of these with calls to .x(),.y() etc member functions to make it more portable--

 I'm not using arm neon, but it appears there is vgetq_lane_f32 for extraction, and vsetq_lane_f32 for setting.

I made this change to my version of Bullet, but its it has other changes so I wanted to see if this was something Erwin(or whatever makes these decisions) was okay with before submitting anything(since I'd have to redo it in Bullets master branch).

gjaegy, 2019-07-04:
That's a very interesting finding in my opinion. 

Have you received any feedback from Erwin ? 

I think your suggestion should be implemented since providing a significant performance win (I'm also trying to speed up bullet a bit through AVX - we use the double-precision version here)

TrianglesPCT, 2019-07-09:
Hi gjaegy, I haven't heard anything about this topic.

Bullet is probably mostly memory  bound, so without a large rewrite to make the entire thing much more SoA friendly it probably doesn't make a huge difference anyway.

gjaegy, 2019-07-10:
Yes, I think you are right, I came to the same conclusion after having done that AVX code path implementation :/

closed 2021-04-26 by erwincoumans