bulletphysics/bullet3 #2084: Ball Motion Trajectory

irwantozn1, 2019-01-29:
Hello guys, I'm currently on the project to create football game using bullet physics. All objects (e.g. field, player, ball) are using bullet body, so all contacts calculated by this library . Currently I found 2 problems:
1. Sometimes the ball will fail to detect collision on these conditions:
   a. Ball moving and should collide with goal net, but the ball sometimes penetrating to goal net
   b. When the ball is dribbling or moving on the ground, sometimes it will falling down penetrate the field

How to solve this?

2. I want to create prediction of ball trajectories before shoot. I'm ever trying to call world->stepSimulation(...) on main update, but it will make game freeze for a while,, is there any suitable method to create this trajectory ? or How to call all simulation process but only for ball ?

erwincoumans, 2019-01-31:
for (1), likely the timestep is too big, for the ball size/speed, use a smaller timestep.

for (2), it is easiest to create a second (or more) dynamics world, do the ball simulation/prediction in there.

closed 2019-01-31 by erwincoumans

irwantozn1, 2019-02-08:
(2) Thanks for your suggestion to using second world (world A which has all bodies, world B to create array), currently I can create an array without stopping the game.

(1) Did you mean fixedTimeStep or timeStep ?
fixedTimeStep (m_fixedTimeStep) still using  1/60.f as suggested on this library, but The time step is using system frame dt, it's range on 0.01 ~ 0.05 s. Currently I'm using m_useContinuous = true and additional customize prediction using raycast on main thread, because using one of them still found penetrating issue with very rare frequency.

But I thought that I found new problem after use continuous. The ball sometimes going to wrong direction after collide and bouncing on the ground even thought there's no object blocked the ball. What are the possible causes of this ?

Here I attaching a file:

[image]

The red line is ball trajectory which computed using world B, collision objects are ball, field, goal nets. White collision is ghost object on world A (currently I debugdraw objects on world A).

* I've tried to check colObj0 & colObj1 on btSequentialImpulseConstraintSolver::convertContact(...) and it always ball and stadium (I see from their worldtransform) it should not caused by colliding with goal net (invalid position).

Thanks for any help

erwincoumans, 2019-02-08:
Use a smaller fixed timestep, for example 120Hz or even 240hz. I would not use 'continuous' ccd, since it may interfere with restitution/bounce.

irwantozn1, 2019-02-10:
I'm already change time step and use_continuous = false. Unfortunately this still hapend, I'm trying to compare the m_linearVelocity direction before & after added by m_deltaLinearVelocity. I print data on log file below, this ended on first turning if >5 degrees. Here I print log file:

[debug_wrong_dir2.txt](https://github.com/bulletphysics/bullet3/files/2848569/debug_wrong_dir2.txt)

As we seen that m_deltaLinearVelocity changed by _gResolveSingleConstraintRowLowerLimit_sse4_1_fma3_ and _gResolveSingleConstraintRowLowerLimit_sse4_1_fma3_

I couldn't see how the method works, but it seems like caused by invalid contact normal direction (it should always 0,0,1 since the ball only collide with ground). And it makes the ball direction's turning about 74 degrees

[image]

or you can see on the last log print below:

_SOLVER BODY id - 0
delta_vel :   0.0000 |   0.0000 |   0.0000 | btSolverBody::writebackVelocityAndTransform
deg - delta_vel v.s. prev_vel :  -1.#IND | btSolverBody::writebackVelocityAndTransform
deg - cur_vel v.s. prev_vel :   0.0000 | btSolverBody::writebackVelocityAndTransform
SOLVER BODY id - 1
deg - tran_vel v.s. ang_vel :  -1.#IND | btSolverBody::getVelocityInLocalPointNoDelta
deg - delta_vel v.s. lin_vel :  -1.#IND | btSolverBody::internalApplyImpulse
delta_vel :   0.0000 |   0.0000 |   0.0000 | btSolverBody::internalApplyImpulse
linear comp :  -0.3176 |   0.2194 |  -0.3177 | btSolverBody::internalApplyImpulse
impulse mag :   0.0000 | btSolverBody::internalApplyImpulse
deg - tran_vel v.s. lat_dir :   8.6180 | btSequentialImpulseConstraintSolver::convertContact
deg - delta_vel v.s. lin_vel :  -1.#IND | btSolverBody::internalApplyImpulse
delta_vel :   0.0000 |   0.0000 |   0.0000 | btSolverBody::internalApplyImpulse
linear comp :   0.3702 |   0.0561 |  -0.3313 | btSolverBody::internalApplyImpulse
impulse mag :   0.0000 | btSolverBody::internalApplyImpulse
deg - tran_vel v.s. ang_vel :  61.5673 | btSolverBody::getVelocityInLocalPointNoDelta
deg - delta_vel v.s. lin_vel :  -1.#IND | btSolverBody::internalApplyImpulse
delta_vel :   0.0000 |   0.0000 |   0.0000 | btSolverBody::internalApplyImpulse
linear comp :   0.0000 |   0.0000 |   0.5000 | btSolverBody::internalApplyImpulse
impulse mag :   0.0000 | btSolverBody::internalApplyImpulse
deg - tran_vel v.s. lat_dir :   0.0000 | btSequentialImpulseConstraintSolver::convertContact
deg - delta_vel v.s. lin_vel :  -1.#IND | btSolverBody::internalApplyImpulse
delta_vel :   0.0000 |   0.0000 |   0.0000 | btSolverBody::internalApplyImpulse
linear comp :   0.5000 |   0.0000 |   0.0000 | btSolverBody::internalApplyImpulse
impulse mag :   0.0000 | btSolverBody::internalApplyImpulse
before FMADD delta_vel body 1:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowLowerLimit_sse4_1_fma3
before FMADD delta_vel body 2:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowLowerLimit_sse4_1_fma3
after delta_vel body 1:  -6.4669 |   4.4676 |  -6.4693 | gResolveSingleConstraintRowLowerLimit_sse4_1_fma3
after delta_vel body 2:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowLowerLimit_sse4_1_fma3
contact normal 1:  -0.6353 |   0.4389 |  -0.6355 | gResolveSingleConstraintRowLowerLimit_sse4_1_fma3
contact normal 2:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowLowerLimit_sse4_1_fma3
before FMADD delta_vel body 1:  -6.4669 |   4.4676 |  -6.4693 | gResolveSingleConstraintRowLowerLimit_sse4_1_fma3
before FMADD delta_vel body 2:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowLowerLimit_sse4_1_fma3
after delta_vel body 1:  -6.4669 |   4.4676 |  31.6548 | gResolveSingleConstraintRowLowerLimit_sse4_1_fma3
after delta_vel body 2:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowLowerLimit_sse4_1_fma3
contact normal 1:   0.0000 |   0.0000 |   1.0000 | gResolveSingleConstraintRowLowerLimit_sse4_1_fma3
contact normal 2:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowLowerLimit_sse4_1_fma3
before FMADD delta_vel body 1:  -6.4669 |   4.4676 |  31.6548 | gResolveSingleConstraintRowGeneric_sse4_1_fma3
before FMADD delta_vel body 2:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowGeneric_sse4_1_fma3
after FMADD delta_vel body 1: -10.9895 |   3.7821 |  35.7024 | gResolveSingleConstraintRowGeneric_sse4_1_fma3
after FMADD delta_vel body 2:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowGeneric_sse4_1_fma3
contact normal 1:   0.7405 |   0.1122 |  -0.6627 | gResolveSingleConstraintRowGeneric_sse4_1_fma3
contact normal 2:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowGeneric_sse4_1_fma3
before FMADD delta_vel body 1: -10.9895 |   3.7821 |  35.7024 | gResolveSingleConstraintRowGeneric_sse4_1_fma3
before FMADD delta_vel body 2:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowGeneric_sse4_1_fma3
after FMADD delta_vel body 1: -24.8001 |   3.7820 |  35.7024 | gResolveSingleConstraintRowGeneric_sse4_1_fma3
after FMADD delta_vel body 2:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowGeneric_sse4_1_fma3
contact normal 1:   1.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowGeneric_sse4_1_fma3
contact normal 2:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowGeneric_sse4_1_fma3
before FMADD delta_vel body 1: -24.8001 |   3.7820 |  35.7024 | gResolveSingleConstraintRowLowerLimit_sse4_1_fma3
before FMADD delta_vel body 2:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowLowerLimit_sse4_1_fma3
[... cut ...]
contact normal 1:   1.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowGeneric_sse4_1_fma3
contact normal 2:   0.0000 |   0.0000 |   0.0000 | gResolveSingleConstraintRowGeneric_sse4_1_fma3
SOLVER BODY id - 0
**delta_vel : -29.5027 |  23.2837 |  31.0325 | btSolverBody::writebackVelocityAndTransform**
deg - delta_vel v.s. prev_vel : 141.7193 | btSolverBody::writebackVelocityAndTransform
deg - cur_vel v.s. prev_vel :  **74.5347** | btSolverBody::writebackVelocityAndTransform
DIRECTION IS CHANGE ??? btSolverBody::writebackVelocityAndTransform_

erwincoumans, 2019-02-10:
Is your ground plane a btPlaneShape, a single large btBoxShape?
If it is a triangle mesh or multiple ground shapes, the ball may be hitting an internal edge.
You should be able to add a contact added callback, and verify the contact normal.
For a triangle mesh, you can adjust the contact normal for when the ball hits an internal edge, using some utility:

~~~~~~~~~~~~~~~~~~~~~
#include "BulletCollision/CollisionDispatch/btInternalEdgeUtility.h"

bool MyContactAddedCallback(btManifoldPoint& cp, const btCollisionObjectWrapper* colObj0Wrap, int partId0, int index0, const btCollisionObjectWrapper* colObj1Wrap, int partId1, int index1)
{
	btAdjustInternalEdgeContacts(cp, colObj1Wrap, colObj0Wrap, partId1, index1);
	return true;
}

{ 
//somewhere in your initialization
gContactAddedCallback = MyContactAddedCallback;
}
~~~~~~~~~~~~~~~~~~~~~

irwantozn1, 2019-02-11:
Thanks for your help, I thought the default value of m_useContinuous is false, so I just comment my declaration world->m_useContinuous = true; but unfortunately it's default was true, so I change my declaration to world->m_useContinuous = false; . As you said, this fix the problem, but I have a last question for this issue.

Why on convexSweepTest by this continuous detection was not calling gContactAddedCallback ?

I think if it's calling gContactAddedCallback will fix this issue without make continuous to false.

irwantozn1, 2019-02-12:
Thanks for your help, currently the bug was fixed by add collision callback and forced the contact normal on the field to 0,0,1.

Regards,