bulletphysics/bullet3 #2085: Possible raytracing miscalculation in createVisualShape.py in macOS with retina screen.

WNoxchi, 2019-01-30:
PyBullet pip-installed 2019-01-29; macOS 10.14.2

In `bullet3/examples/pybullet/examples/createVisualShape.py` the color of an object doesn't change when it's clicked on. Instead it seems to be offset left and upwards: if you click towards the bottom right of the window you can change an object's color. This offset appears to move back towards the center as you zoom in.

Here is a recording demonstrating the issue:
https://youtu.be/8M9bb05K-zk

I don't know how to fix this at this time so I'm bringing it to attention. My guess is this is a retina screen issue: I ran `createVisualShape.py` on a Linux computer and it worked normally. Perhaps something somewhere needs to be multiplied or divided by `2`.

erwincoumans, 2019-01-30:
Yes, the you can multiply the mouse coords by 2 for now. We need to compensate the 'vertical' and 'horizontal' with that factor.

WNoxchi, 2019-01-30:
Oh that does work, tested it now. I could do a PR checking for macOS with `sys.platform == 'darwin'`; not sure if you want that. I'm also unsure whether that's a good check: modern Macs use retina screens, but maybe some and the older ones don't?

erwincoumans, 2019-01-30:
Fixed here: https://github.com/bulletphysics/bullet3/pull/2087

closed 2019-01-31 by erwincoumans