bulletphysics/bullet3 #2089: Is the SDF Collision Shape ready for use yet?

devshgraphicsprogramming, 2019-01-31:
I was trying to investigate the feasibility of integrating Bullet into an engine with voxel terrain, so far I've came up  with 3 approaches:
1) Implement my own Octree Collision Shape (re-use the octrees data I already have)
2) Overload and derive from one of your BVH Shape (since your bvh is a tree of boxes) classes to get an octree 
3) Use Signed Distance Fields

While examining the source I stumbled across the btSDFCollisionShape class, but it had relatively little documentation and was only added 9 months ago, hence my question.

Is it ready to use for static (mass=0) colliders?
Also are there plans to allow the user to provide their own MiniSDF implementation?
It would be awesome to allow the reuse of the conservative approximation of the SDF that many engines use for sphere-tracing shadows on the GPU.

erwincoumans, 2019-02-01:
How many boxes/voxels do you have? 

Signed distance fields are not a good approximation for boxes, and the memory usage is way to large to be practical.

I would use the dynamic AABB tree or create a custom acceleration structure.

closed 2019-02-01 by erwincoumans

devshgraphicsprogramming, 2019-02-01:
> How many boxes/voxels do you have?

Pages of 256 thousand (could be made smaller), but they are all static.

> Signed distance fields are not a good approximation for boxes, and the memory usage is way to large to be practical.

I would be tempted to agree, but I already have an SDF and an octree for other purposes.
When you say "not a good approximation" do you mean that SDF is treated like a level surface, i.e. that the normals and interfaces are defined for a smooth surface where distance=0.0 ?
That could actually be ideal for terrain (or anything else) generated with marching cubes from exactly that representation.

> I would use the dynamic AABB tree or create a custom acceleration structure.

That would be `btDbvtBroadphase` correct? I was more after a static collision shape that I could use to have less rigid bodies in the scene.

erwincoumans, 2019-02-01:
If you already have an sdf that represents the static shape, that may work. There is a PyBullet demo showing sdf collisions. What format is your SDF? Is the code public?

btCompoundShape uses a btDbvt as dynamic aabb tree, you could try that.  

btDbvtBroadphase uses the same btDbvt.