bulletphysics/bullet3 #2092: pybullet removeBody: Don't remove collision shape: it is used

kyktommy, 2019-02-01:
pybullet when i call removeBody to remove a body, then add the body in the same position, it shows following message, and the application is dead. 

```
# b3Printf: Don't remove collision shape: it is used.
# b3Printf: b3Warning[examples/SharedMemory/PhysicsServerCommandProcessor.cpp,8964]:
```

I have tried some other method, but still not works, e.g: 
- move the body to other position before remove
- changeDynamics mass=0 before remove
- p.setCollisionFilterGroupMask(obj, -1, 0, 0) before remove

Why i got this message when remove body?

erwincoumans, 2019-02-01:
Please provide a small self-contained reproduction script, preferably no dependencies aside from pybullet. Thanks!

erwincoumans, 2019-02-01:
Attached is a working example. Please modify and re-open if you reproduce the issue.
[removeBody.zip](https://github.com/bulletphysics/bullet3/files/2820146/removeBody.zip)

closed 2019-02-01 by erwincoumans