bulletphysics/bullet3 #2096: get base URDF link frame position problem

xinsongyan, 2019-02-04:
In pybullet,  getBasePositionAndOrientation() gives back the position of CoM of base link, when the URDF base link frame is needed, I have to do local transformation by myself. 

The getLinkState() could be another way to get the URDF base link frame information, but it seems that link index -1 is not accepted (for some other function it stands for base link), What is the consideration behind this?

rubdub, 2019-03-13:
I had this issue as well. getLinkState(robotId, -1, 1) gives an error

erwincoumans, 2019-03-13:
Yes, getLinkState only works for child links that are connected by a joint to the parent.

For the base, you need to call getBasePositionAndOrientation, which provides the position and orientation of the center of mass. You can use getDynamicsInfo(bodyUniqueId,-1) to get more information about the base, such as the localInertialFrame. Then use pybullet.multiplyTransforms to get the link transform of the base.

closed 2019-03-13 by erwincoumans