bulletphysics/bullet3 #2100: Springs in pybullet

krasin, 2019-02-06:
I would like to use springs to define a complex body that consists of multiple parts. In the C++ land, there's btGeneric6DofSpring2Constraint. Unfortunately, I can't find a way to use it from pybullet.

Would it be a correct guess that pybullet is missing plumbing for this constraint? How hard it would be to add?

I am willing to take on that, but will appreciate some guidance, as I am relatively new to the Bullet / pybullet code.

Thank you in advance.

erwincoumans, 2019-02-07:
By default, PyBullet uses btMultiBody reduced coordinate articulated bodies. 
 btGeneric6DofSpring2Constraint is only implemented for btRigidBody.
You can loadURDF using the useMaximalCoordinates=True flag, and in that case, joints are implemented as  btGeneric6DofSpring2Constraint. So we could add some API that lets you configure existing btRigidBody joints that way. 

Notice that useMaximalCoordinates is not well supported in PyBullet, since most users use the default reduced coordinate methods.

krasin, 2019-02-07:
Hi Erwin,

thank you very much for your throughout reply.

Using useMaximalCoordinates and exposing necessary APIs might be the way to go, and I would be happy to code that, but before we dive into the implementation details, let me take a step back and explain the higher-level issue I am trying to solve.

We aim to learn how to optimize gripper shapes and for that we will need to have a better idea which parts of the gripper are affected by the contact when an object is grasped. It's great that pybullet has getContactPoints, but it will only give surface-level contact points. We would like to have deeper insights and the approach we currently explore is to make grippers out of voxels connected with pretty stiff springs organized in a cubic crystal.

That way, after an object is grasped, the points of immediate contact will get highest displacement values, while their neighbors will get lower, and the voxels which don't participate in the contact at all will get ~zeros.

Given that we seem to be outside of the regular pybullet use cases, I wonder if the same problem could be solved by other means. Realistically, we don't care if it's displacements or some other measure of importance for a specific contact event. What we care is that this information is spreaded among higher number of voxels so that we can make better decisions on how to optimize the shape.

Let me know, if that makes sense to you. :)

erwincoumans, 2019-02-07:
How about using a fixed constraint with a very limited force? That will act like a spring, and you can read the applied forces. This works with regular reduced coordinates.
See attached example. You can add a whole bunch of spheres/boxes next to each other, and use fixed constraint (PyBullet.createConstraint) to keep them there. Then use PyBullet.changeConstraint(cid,maxForce=...) to limit the constraint force.
[constraint.zip](https://github.com/bulletphysics/bullet3/files/2842913/constraint.zip)

I think the [fixed](https://github.com/bulletphysics/bullet3/blob/master/src/BulletDynamics/Featherstone/btMultiBodyFixedConstraint.cpp) constraint also works between reduced and maximal coordinate objects, so you can use maximal coordinate spheres for the surfaces and reduced coordinate for the gripper.

krasin, 2019-02-08:
Hi Erwin,

that sounds really great! I will spend some time tomorrow to see how many voxels it will be possible to simulate with a reasonable simulation speed.

Will report back.

erwincoumans, 2019-02-08:
It sounds a bit like this (thanks Yunfei Bai for the link)
https://www.cc.gatech.edu/graphics/projects/Sumit/homepage/projects/softcontacts/index.html

erwincoumans, 2019-02-08:
Assuming this works, let's close the issue. Please re-open if not.

closed 2019-02-08 by erwincoumans

horuswoo, 2020-10-14:
Is it possible to add a spring (possibly implemented with a bunch of spheres/boxes with constraint, like @erwincoumans mentioned above) as a convenience feature? Also wondering if @krasin was able to get it working, as I'm seeing other issues following this approach facing other issues like #3047