bulletphysics/bullet3 #2101: fingertips of gripper moving with inertia

lukashermann, 2019-02-06:
Hello everyone,
I'm using pybullet to simulate a robotic parallel gripper with two prismatic joints (one each finger). I encountered the problem that either:
- the force for the prismatic joints is too high and the physical interaction with the grasping object is not modelled correctly (e.g. finger entering surface of the object)
OR 
- the force is lower and the grasp is simulated correctly, but then the fingers dont stay in position as the robot moves the gripper from side to side (I guess because of inertia).

I already tried to lower the finger's mass and inertia and played around with the force, but I dont achieve a satisfying result as either of the unfavorable conditions described above holds. Which parameters could I try to change?

ErickKramer, 2020-01-12:
Were you able to find a solution to the floating fingers problem?. I am also facing similar issues especially when the fingers collide with an object.

erwincoumans, 2020-01-12:
The forces of both fingers against the object cancel eachother out perfectly, so there is no force that keeps the fingers centered. One solution is to add a gear constraint to explicitly keep the fingers centered:
https://www.youtube.com/watch?v=yKShjSTayco
Also, finger forces should not be too large, otherwise penetration will happen and you never get out of penetration. Use a smaller timestep if you really need such large forces.
Also, friction_anchor is useful to add, to avoid the object slipping out of the grasp. In addition, some amount of spinning_friction (around the contact normal) will reduce the object rotating in the grasp.
All of this is implemented in this example:
https://github.com/bulletphysics/bullet3/tree/master/examples/pybullet/gym/pybullet_robots/panda

closed 2020-01-12 by erwincoumans

ErickKramer, 2020-01-12:
Thanks for the reply! @erwincoumans . I will take a look into the example you provided and try to solve my problem.

ErickKramer, 2020-01-14:
I just want to point out that adding a [gear constraint] (https://github.com/bulletphysics/bullet3/blob/6428ed5e5f0641321dfbffe29b32a7fd423ca9c2/examples/pybullet/gym/pybullet_robots/panda/panda_sim_grasp.py#L46) to the gripper solved the issue of having loose fingers when colliding with an object.

Alchemist77, 2020-02-13:
Hello,

I applied all variable from erwincoumans. Finally, the gripper is stronger than before. However, after my robot (UR5 + robotiq140) grasp a block, the gripper slightly open during the robot arm lift the object then finally the robot drops the object.

Do you know how to fix the problem? 
My robot moves up (z-axis) 1 cm each step and I added time.sleep (0.005) otherwise, the robot cannot be stable in rendering.
In the video, how to keep the object even though the arm life up at least 10 cm??

Thank you.

ErickKramer, 2020-02-13:
@Alchemist77 

> I applied all variable from erwincoumans. Finally, the gripper is stronger than before. However, after my robot (UR5 + robotiq140) grasp a block, the gripper slightly open during the robot arm lift the object then finally the robot drops the object.
> Do you know how to fix the problem?

Perhaps adding lateral friction to the fingers of the gripper might help (I defined it in the URDF of the arm). If not, increasing the force could also be a possible solution.

Alchemist77, 2020-02-13:
Thank you for suggestion!
I already did in code and urdf. 
I found that if I apply this line,
p.changeConstraint(c, gearRatio=-1, erp=0.1, maxForce=50)
it can happen like open gripper...

Still, I am not sure that when i give the same position then sometimes the robot can grasp an object and sometimes it doesn't...

Alchemist77, 2020-02-13:
Then i use this code (add lateral friction directly to the block)
blockStartPos = [0.1, -0.49, 0.87]
blockStartOrientation = p.getQuaternionFromEuler([0, 0, 0])
boxId = p.loadURDF("./urdf/objects/block.urdf", blockStartPos, blockStartOrientation, flags = option_flag)
p.changeDynamics(boxId, -1,lateralFriction = 5,linearDamping = 0,angularDamping = 0)

Alchemist77, 2020-02-13:
Is there another way to add friction?? I added friction in URDF and code with this function (p.changeDynamics), but still the gripper missed the object even though the gripper full grasped the object...