bulletphysics/bullet3 #2107: pybullet_envs not rendering

BruceK4t1qbit, 2019-02-12:
Here's my code

[image]

BruceK4t1qbit, 2019-02-12:
It does work if I add the env.render() before the env.reset(), rather than each time I want to render.

Is this the intended behaviour?

[image]

erwincoumans, 2019-02-12:
Yes, that is the intended behavior. Problem is that Gym doesn't have a boolean render when creating the environment. Bullet needs to enable rendering at initialization, not afterwards.

It requires a lot of changes, to allow opening/closing a render window at any time, so for now, using the 'render' before first reset is needed.

closed 2019-02-12 by erwincoumans

BruceK4t1qbit, 2019-02-13:
Can I add a check, or a print statement to this effect?  To save the next person a couple hours?