bulletphysics/bullet3 #2110: getCameraImage with openGL returns empty image in VR (shared memory)

staehlii, 2019-02-15:
Hello!

I am using pybullet to record data with the VR feature. I connect to the VR server just like it is explained in the documentation and it works well. 
However when I use getCameraImage with openGL enabled to get the images from the VR they are empty. It does work with tinyrenderer but I can not use it because the performance drops too much.

The easiest way to test this is with the [(testrender.py).](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/testrender.py) that you already have  By changing the connection to `pybullet.SHARED_MEMORY` and having the VR server running you will see that the images are blank.

Is there an option to get the openGL cameras from the VR server?

Thanks and regards,
Ian

borsuk74, 2019-02-17:
I hit the same problem in this scenario too , from debugger observed that  call to:
glReadPixels(0, 0, width, height, GL_RGBA, GL_UNSIGNED_BYTE, rgbaBuffer)
fails with error code  GL_INVALID_OPERATION. 
I have OpenGL 4.6 installed on Windows 10, renderer is GeForce GTX 1070 .

BluePrintRandom, 2019-02-17:
Wmr Mixed reality headset recently changed to a new USB standard for the
image stream,

erwincoumans, 2019-02-18:
You hit an unimplemented path indeed, should be easy to implement.

sholtodouglas, 2019-04-07:
@BluePrintRandom  what implications does that have?
@borsuk74, your fix stopped the program crashing when I get images with openGL, but now I'm getting blank images as above. Is it currently working well for you?

erwincoumans, 2019-04-14:
Thanks for sharing a fix.
Have you considered saving/setting/restoring the framebuffer outside of the renderer,
so you don't need to duplicate the renderScene into renderSceneInBuffer?

erwincoumans, 2019-04-14:
Here is a one-line fix:
https://github.com/bulletphysics/bullet3/commit/219970c09eeae823f7dc8a7488e58b33917db9b5
Thanks for the report!

closed 2019-04-14 by erwincoumans