bulletphysics/bullet3 #2114: Build issue -- argument value outside of valid range

collenjones, 2019-02-18:
OS: macOS 10.14.2
Compiler: Clang 7.0.1

I'm still relatively new to C++ so it's likely I'm doing something dumb but I'm getting the following error when I run `./build_cmake_pybullet_double.sh`:

```
/Users/collen/Dev/Language/C++/bullet3/src/Bullet3Common/b3Vector3.h:290:7: error: argument value 10880 is outside the valid range [0, 255]
      [-Wargument-outside-range]
```

Any ideas how to fix this?

# Full Trace

```
./build_cmake_pybullet_double.sh
-- The C compiler identification is Clang 7.0.1
-- The CXX compiler identification is Clang 7.0.1
-- Check for working C compiler: /usr/local/opt/llvm/bin/clang
-- Check for working C compiler: /usr/local/opt/llvm/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/local/opt/llvm/bin/clang++
-- Check for working CXX compiler: /usr/local/opt/llvm/bin/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Apple
-- Found OpenGL: /System/Library/Frameworks/OpenGL.framework
OPENGL FOUND
/System/Library/Frameworks/OpenGL.framework/System/Library/Frameworks/OpenGL.framework
-- Found PythonInterp: /usr/local/bin/python3 (found version "3.7.2")
-- Looking for versions: 3.7.2;3.7
-- Looking for python version '3.7.2' by checking executables: /usr/local/bin/python3;python;python3;python3.7.
-- Found executable /usr/local/bin/python3 with suitable version 3.7.2
-- Found PythonLibs: /usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/lib/libpython3.7.dylib (found suitable exact version "3.7.2")
-- Found PythonLibs: /usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/lib/libpython3.7.dylib (found version "3.7.2")
-- Found NumPy: /usr/local/lib/python3.7/site-packages/numpy/core/include (found version "1.16.1")
-- NumPy ver. 1.16.1 found (include: /usr/local/lib/python3.7/site-packages/numpy/core/include)
NumPy found
Mac OSX Version is 10.14
/System/Library/Frameworks/Cocoa.framework
/System/Library/Frameworks/Cocoa.framework
/System/Library/Frameworks/Cocoa.framework
/System/Library/Frameworks/Cocoa.framework
CL64
/System/Library/Frameworks/Cocoa.framework
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    BT_USE_EGL

-- Build files have been written to: /Users/collen/Dev/Language/C++/bullet3/build_cmake
Scanning dependencies of target Bullet3Common
Scanning dependencies of target BussIK
Scanning dependencies of target clsocket
Scanning dependencies of target gtest
Scanning dependencies of target HACD
Scanning dependencies of target LinearMath
Scanning dependencies of target gwen
[  0%] Building CXX object examples/ThirdPartyLibs/clsocket/CMakeFiles/clsocket.dir/src/SimpleSocket.o
[  0%] Building CXX object Extras/HACD/CMakeFiles/HACD.dir/hacdGraph.o
[  1%] Building CXX object examples/ThirdPartyLibs/clsocket/CMakeFiles/clsocket.dir/src/ActiveSocket.o
[  1%] Building CXX object src/Bullet3Common/CMakeFiles/Bullet3Common.dir/b3AlignedAllocator.o
[  1%] Building CXX object examples/ThirdPartyLibs/BussIK/CMakeFiles/BussIK.dir/Jacobian.o
[... 101 lines cut ...]
[  9%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Controls/TextBox.o
[  9%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Controls/TextBoxNumeric.o
[  9%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Controls/TreeControl.o
[  9%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Controls/TreeNode.o
[  9%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Controls/VerticalScrollBar.o
[  9%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Controls/VerticalSlider.o
[  9%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Controls/WindowControl.o
[  9%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/DragAndDrop.o
[  9%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Gwen.o
[  9%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Hook.o
[  9%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Platforms/Windows.o
[ 10%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Platforms/Null.o
[ 10%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Renderers/OpenGL_DebugFont.o
[ 10%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Skin.o
[ 10%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/ToolTip.o
[ 10%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Utility.o
[ 10%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/events.o
[ 10%] Building CXX object examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/inputhandler.o
[ 10%] Linking CXX shared library libgwen.dylib
ld: warning: text-based stub file /System/Library/Frameworks//OpenGL.framework/OpenGL.tbd and library file /System/Library/Frameworks//OpenGL.framework/OpenGL are out of sync. Falling back to library file for linking.
ld: warning: text-based stub file /System/Library/Frameworks/OpenGL.framework/Versions/A/Libraries/libGLU.tbd and library file /System/Library/Frameworks/OpenGL.framework/Versions/A/Libraries/libGLU.dylib are out of sync. Falling back to library file for linking.
ld: warning: text-based stub file /System/Library/Frameworks/OpenGL.framework/Versions/A/Libraries/libGL.tbd and library file /System/Library/Frameworks/OpenGL.framework/Versions/A/Libraries/libGL.dylib are out of sync. Falling back to library file for linking.
[ 10%] Built target gwen
make: *** [all] Error 2
```

jamesdolan, 2019-03-30:
Looks like the warning is new when apple updated clang. This should fix it though...
`#define BT_SHUFFLE(x, y, z, w) (((w) << 6 | (z) << 4 | (y) << 2 | (x)) & 0xff)`

collenjones, 2019-05-01:
Thanks, @jamesdolan! This appears to have been fixed so closing.

closed 2019-05-01 by collenjones

erwincoumans, 2019-05-01:
We don't need to fix this on the Bullet source code?

EmilNorden, 2019-05-04:
> We don't need to fix this on the Bullet source code?

I've created a small pull request for this issue: https://github.com/bulletphysics/bullet3/pull/2232

erwincoumans, 2019-05-05:
Thanks a lot for the help everyone!