bulletphysics/bullet3 #2117: Options for ContactSolverMode in pybullet

FieldMrFive, 2019-02-19:
Trying to build a pool game simulation. 
**Problem**: The ball will always have a curved trajectory. I have checked the velocities, the angle between directions of linear velocity and rotation velocity is perfect 90 degree (in x y planar), so theoretically it must roll in straight. But the linear velocity's direction keeps changing.

Similar problem here suggests change ContactSolverMode, but I am using pybullet and it does not have the API: https://pybullet.org/Bullet/phpBB3/viewtopic.php?t=10476.

Or other ways to solve it are also welcome.

erwincoumans, 2019-02-19:
Please provide a small reproduction case as a PyBullet file that shows the issue.

FieldMrFive, 2019-02-19:
> Please provide a small simple reproduction case as a PyBullet file that shows the issue (you can attach a zip file to this issue).

Here is a demo. Run `pool_env_debug.py`.
As you can see the cue ball ends up going strictly parallel to x axis, which is problematic.

[pool.tar.gz](https://github.com/bulletphysics/bullet3/files/2881081/pool.tar.gz)

erwincoumans, 2019-02-19:
Thanks for sharing. Can you help isolating the issue further?

It would be good to keep things a bit more simple.
1) set rolling friction to zero
2) set spinning friction to zero
3) remove cfm, erp, stiffness, damping and other contact parameters from URDF
4) set linear and angular damping to 0
5) use an infinite plane instead of the pool table
6) use some initial velocity for the ball
7) the velocity of the ball should stay constant

See balltest.py in attached example.
[tmp.zip](https://github.com/bulletphysics/bullet3/files/2881505/tmp.zip)

From this simple example, make one modification at a time, and see what makes the trajectory bend.

By the way, the ball shouldn't slide. For sliding boxes, there is a cone friction, but this doesn't apply:
~~~~~~
 p.setPhysicsEngineParameter(enableConeFriction=1)
~~~~~~

FieldMrFive, 2019-02-19:
@MichaelMcDonnell 
First I use a big plane, setting the contact properties as the pool table. And now I set the initial linear velocity instead of applying force.
Trajectories are still bend, but I found some interesting cases: If the velocity's direction is 0° 45° 90° 135° 180° 225° 270° 315°, the trajectories will be straight, otherwise bend.

[image]
[image]

Then, after setting both planar and ball's rolling friction = 0, ball rolls straight. I have also tried set only spinning friction = 0 or only lateral friction = 0, bend is still there. Setting linear damping and angular damping, it's still there.
So I assume it is something with rolling friction, so I give only rolling friction in URDFs, the trajectories bend.
Also, the trajectory will bend to parallel itself to the near axis asymptotically. Initial velocity (0°, 45°), it will bend to x axis; Initial velocity (45°, 90°), it will bend to y axis; etc.

Test code: 
[pool-debug.zip](https://github.com/bulletphysics/bullet3/files/2881989/pool-debug.zip)

FieldMrFive, 2019-02-19:
Regarding sliding of balls, one purpose of using physics engines is to simulating the sliding and rolling mixture of billiard.

erwincoumans, 2019-02-19:
>> So I assume it is something with rolling friction,

Yes, rolling friction can cause the bending indeed. We added cone friction for lateral friction. Instead of clamping individual friction directions, cone friction clamps the combined friction vector. See the [coneFriction](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/frictionCone.py) for example.

We can implement such combined friction handling for rolling friction as well.

FieldMrFive, 2019-02-20:
OK, when will it probably be done? This is the environment I used for my master thesis, kinda urgent, can't train anything without it.

Well currently I think of a workaround, just set rolling friction = 0, and apply manually external force based on contact normal force and angular velocity direction. For this simple case of pool - moving on a plane, it might just work.
I will try and find out.

erwincoumans, 2019-02-26:
Indeed best to go for a workaround, we lack resources to help all PyBullet users.

erwincoumans, 2019-03-04:
OK, I got around implementing implicit cone friction for rolling friction:

[image]

By the way, you may also want to add some spinning friction for the ball, so if you set some angular velocity, it will be reduced as well.
~~~~~~~~~
 <contact>
      <rolling_friction value="0.00001"/>
      <spinning_friction value="0.00001"/>
    </contact>
 p.resetBaseVelocity(ball, [0,0,0], [0, 0, 1])
 v = p.getBaseVelocity(ball)[1]
        print("vel ", v)
~~~~~~~~~

erwincoumans, 2019-03-04:
@FieldMrFive can I add the pool table to PyBullet for some tests/examples? If yes, under a permissive license, such as zlib/mit/bsd etc?

FieldMrFive, 2019-03-04:
@erwincoumans Yes, of course, just select any opensource license.
And thank you for implementing for cone friction of rolling friction.

erwincoumans, 2019-12-23:
Rolling friction cone was implemented long ago for btMultiBody (NOT btRigidBody)
(at least the sphere with rolling frition enabled needs to be btMultiBody, the plane/other geometry can be btRigidBody or btMultiBody)
Also best to enable this patch, so rolling friction is enabled for all contact points:
https://github.com/bulletphysics/bullet3/pull/2553

PyBullet example attached and images
[image]
[image]
[frictionCone.zip](https://github.com/bulletphysics/bullet3/files/3995940/frictionCone.zip)

closed 2019-12-23 by erwincoumans

rolfvdhulst, 2019-12-23:
Amazing, thanks!