bulletphysics/bullet3 #2118: Does Evolution in examples was right codes?

Evi5, 2019-02-19:
I am a newer and try to learn with demos, and maybe that I am wrong and the bullet is well done.

The example named 'Neural Network 3D Walkers' under Evolution in App_ExampleBrowser wan not get the best Walker in 10 minutes， the result was disorganized.

I make some change in NN3DWalkers.cpp, let walkers in different position and print their angle when they fall down from stand with motor does nothing.

    void NN3DWalkersExample::scheduleEvaluations()
    {
        //...
        m_walkersInPopulation[i]->resetAt(btVector3(0, 0, 4*i));
        //...
    }
    void NN3DWalkersExample::updateEvaluations(const btScalar timeSinceLastTick)
    {
        //...
		for (int r = 0; r < NUM_WALKERS; r++)
		{
			if (m_walkersInPopulation[r]->isInEvaluation())
			{

				btHingeConstraint* hingeA = static_cast<btHingeConstraint*>(m_walkersInPopulation[r]->getJoints()[0]);
				btHingeConstraint* hingeB = static_cast<btHingeConstraint*>(m_walkersInPopulation[r]->getJoints()[1]);
				btHingeConstraint* hingeC = static_cast<btHingeConstraint*>(m_walkersInPopulation[r]->getJoints()[2]);
				btHingeConstraint* hingeD = static_cast<btHingeConstraint*>(m_walkersInPopulation[r]->getJoints()[3]);
				btHingeConstraint* hingeE = static_cast<btHingeConstraint*>(m_walkersInPopulation[r]->getJoints()[4]);
				btHingeConstraint* hingeF = static_cast<btHingeConstraint*>(m_walkersInPopulation[r]->getJoints()[5]);
				btHingeConstraint* hingeG = static_cast<btHingeConstraint*>(m_walkersInPopulation[r]->getJoints()[6]);
				btHingeConstraint* hingeH = static_cast<btHingeConstraint*>(m_walkersInPopulation[r]->getJoints()[7]);

				printf("%f\t", hingeA->getHingeAngle());
				printf("%f\t", hingeB->getHingeAngle());
				printf("%f\t", hingeC->getHingeAngle());
				printf("%f\t", hingeD->getHingeAngle());
				printf("%f\t", hingeE->getHingeAngle());
				printf("%f\t", hingeF->getHingeAngle());
				printf("%f\t", hingeG->getHingeAngle());
				printf("%f\t\n", hingeH->getHingeAngle());
				continue;
				//...
			}
		}
		printf("\n");
		//...
    }

The output shows they have different behaviors(The 1 to 3 was the same then diff angels), I am confused and stuck on it 2 weeks.
Thks!

> Version = 4.1 INTEL-12.4.7
> Vendor = Intel Inc.
> Renderer = Intel Iris OpenGL Engine
> An evaluation started at 0.016667 s.
> An evaluation started at 0.016667 s.
> An evaluation started at 0.016667 s.
> An evaluation started at 0.016667 s.
> An evaluation started at 0.016667 s.
> An evaluation started at 0.016667 s.
> An evaluation started at 0.016667 s.
> An evaluation started at 0.016667 s.
> An evaluation started at 0.016667 s.
> An evaluation started at 0.016667 s.
> -0.278732	0.219167	-0.272268	0.213556	-0.264524	0.220002	-0.265957	0.215320
> -0.278732	0.219167	-0.272268	0.213556	-0.264524	0.220002	-0.265957	0.215320
> -0.278732	0.219167	-0.272268	0.213556	-0.264524	0.220002	-0.265957	0.215320
> -0.278732	0.219167	-0.272268	0.213556	-0.264524	0.220002	-0.265957	0.215320
> -0.278732	0.219167	-0.272268	0.213556	-0.264524	0.220002	-0.265957	0.215320
> -0.278732	0.219167	-0.272268	0.213556	-0.264524	0.220002	-0.265957	0.215320
> -0.278732	0.219167	-0.272268	0.213556	-0.264524	0.220002	-0.265957	0.215320
> -0.278732	0.219167	-0.272268	0.213556	-0.264524	0.220002	-0.265957	0.215320
> -0.278732	0.219167	-0.272268	0.213556	-0.264524	0.220002	-0.265957	0.215320
> -0.278732	0.219167	-0.272268	0.213556	-0.264524	0.220002	-0.265957	0.215320
> 
> -0.320657	0.215418	-0.318278	0.208701	-0.320839	0.216255	-0.323194	0.210321
> -0.320657	0.215418	-0.318278	0.208701	-0.320839	0.216255	-0.323194	0.210321
> -0.320657	0.215418	-0.318278	0.208701	-0.320839	0.216255	-0.323194	0.210321
> -0.320657	0.215418	-0.318278	0.208701	-0.320839	0.216255	-0.323194	0.210321
> -0.320657	0.215418	-0.318278	0.208701	-0.320839	0.216255	-0.323194	0.210321
> -0.320657	0.215418	-0.318278	0.208701	-0.320839	0.216255	-0.323194	0.210321
> -0.320657	0.215418	-0.318278	0.208701	-0.320839	0.216255	-0.323194	0.210321
> -0.320657	0.215418	-0.318278	0.208701	-0.320839	0.216255	-0.323194	0.210321
> -0.320657	0.215418	-0.318278	0.208701	-0.320839	0.216255	-0.323194	0.210321
> -0.320657	0.215418	-0.318278	0.208701	-0.320839	0.216255	-0.323194	0.210321
> 
> -0.376306	0.217191	-0.382732	0.207463	-0.400049	0.216593	-0.405486	0.209520
> -0.376306	0.217191	-0.382732	0.207463	-0.400049	0.216593	-0.405486	0.209520
> -0.376306	0.217191	-0.382732	0.207463	-0.400049	0.216593	-0.405486	0.209520
> -0.376306	0.217191	-0.382732	0.207463	-0.400049	0.216593	-0.405486	0.209520
> -0.376306	0.217191	-0.382732	0.207463	-0.400049	0.216593	-0.405486	0.209520
> -0.376306	0.217191	-0.382732	0.207463	-0.400049	0.216593	-0.405486	0.209520
> -0.376306	0.217191	-0.382732	0.207463	-0.400049	0.216593	-0.405486	0.209520
> -0.376306	0.217191	-0.382732	0.207463	-0.400049	0.216593	-0.405486	0.209520
> -0.376306	0.217191	-0.382732	0.207463	-0.400049	0.216593	-0.405486	0.209520
> -0.376306	0.217191	-0.382732	0.207463	-0.400049	0.216593	-0.405486	0.209520
> 
> -0.441635	0.221689	-0.463275	0.207912	-0.510368	0.219432	-0.525234	0.211628
> -0.441638	0.221690	-0.463277	0.207912	-0.510367	0.219432	-0.525232	0.211628
> -0.441638	0.221690	-0.463277	0.207912	-0.510367	0.219432	-0.525232	0.211628
> -0.441638	0.221690	-0.463277	0.207912	-0.510367	0.219432	-0.525232	0.211628
> -0.441638	0.221690	-0.463277	0.207912	-0.510367	0.219432	-0.525232	0.211628
> -0.441638	0.221690	-0.463277	0.207912	-0.510367	0.219432	-0.525232	0.211628
> -0.441638	0.221690	-0.463277	0.207912	-0.510367	0.219432	-0.525232	0.211628
> -0.441638	0.221690	-0.463277	0.207912	-0.510367	0.219432	-0.525232	0.211628
> -0.441638	0.221690	-0.463277	0.207912	-0.510367	0.219432	-0.525232	0.211628
> -0.441638	0.221690	-0.463277	0.207912	-0.510367	0.219432	-0.525232	0.211628
> 
> -0.593210	0.228339	-0.582660	0.209974	-0.591658	0.217825	-0.597593	0.205814
> -0.592282	0.229234	-0.582593	0.209963	-0.591552	0.217218	-0.597897	0.205876
> -0.592282	0.229229	-0.582596	0.209962	-0.591553	0.217223	-0.597883	0.205873
> -0.592282	0.229225	-0.582601	0.209961	-0.591554	0.217228	-0.597869	0.205870
> -0.592314	0.229273	-0.582623	0.209967	-0.591549	0.217182	-0.597948	0.205899
> -0.592283	0.229228	-0.582598	0.209962	-0.591554	0.217222	-0.597883	0.205873
> -0.592282	0.229226	-0.582600	0.209961	-0.591554	0.217226	-0.597873	0.205870
> -0.592281	0.229228	-0.582597	0.209961	-0.591553	0.217224	-0.597881	0.205872
> -0.592285	0.229221	-0.582607	0.209960	-0.591555	0.217231	-0.597857	0.205868
> -0.592284	0.229222	-0.582606	0.209960	-0.591555	0.217231	-0.597859	0.205868

erwincoumans, 2019-02-19:
That example was some contribution by @benelot You could ask him?

Evi5, 2019-02-20:
Hi~ @benelot could you help this issues in bullet of evolution example? I can't contact directly, the @ hope worked. thanks ^^

closed 2019-02-26 by erwincoumans

benelot, 2019-03-02:
Hi! Interestingly, being mentioned did not reach me as a notification.Well, here I am. The problem with this demo is the following: It can not optimize since the robots are not properly restored. Their physical settings are not properly restored and thus their performance varies. I think it should only serve as a bullet example that does some simulation. If you can improve it to restore the physics correctly, that is fine, but this might be hard to do if you are new to bullet. It would be best to reimplement it in pybullet, because it features deterministic restoring of robots.