bulletphysics/bullet3 #2119: Camera info doesn't match simulation

lpdon, 2019-02-19:
Hello!

While trying to some simulations and capturing frames using the humanoid_fixed model ("mjcf/humanoid_fixed.xml") I am having the problem that the image captured by the virtual camera doesn't match what I am simulating. Is this some rendering problem related to MCJF? The URDF and SDF models I tested look perfectly fine (for example the KUKA arm in the picture).

[image]

erwincoumans, 2019-02-19:
MJCF support is rudimentary, but its code path is shared with URDF.
Can you provide a simple reproduction case, PyBullet file?

lpdon, 2019-02-19:
```
import pybullet as p
import time
import pybullet_data

physicsClient = p.connect(p.GUI)
p.setAdditionalSearchPath(pybullet_data.getDataPath())
p.setGravity(0,0,-10)
planeId = p.loadURDF("plane.urdf")

kuka = p.loadSDF("kuka_iiwa/kuka_with_gripper.sdf")[0]
p.resetBasePositionAndOrientation(kuka, [1.0, 0, 0], p.getQuaternionFromEuler([0,0,0]))

human = p.loadMJCF("mjcf/humanoid_fixed.xml",flags = p.URDF_USE_SELF_COLLISION_EXCLUDE_ALL_PARENTS)[0]

for i in range (10000):
    try:
      image = p.getCameraImage(200, 200)
      p.stepSimulation()
      time.sleep(1./240.)
    except KeyboardInterrupt:
      break

p.disconnect()
```

This code is already enough to reproduce it. Using the hardware accelerated camera image does however solve the problem and the camera captures the simulation correctly. Thanks a lot!

erwincoumans, 2019-03-03:
Thanks, the regression is fixed now. Still, better to use the OpenGL hardware accelerated camera if possible (or EGL plugin)

closed 2019-03-03 by erwincoumans