bulletphysics/bullet3 #2121: p.getJointStateMultiDof() can't return torques applied in P.POSITION_CONTROL mode

zeshiYang, 2019-02-20:
Hi,every one,
I want to control a spherical joint in robot with PD controller and I hope to get the torque applied on the joint, it should return a list with 3 numbers, but it still return a single float value, how should I cope with this issue?

erwincoumans, 2019-02-25:
Yes, we need to fix this:

https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/pybullet.c#L4321
and
Adding some additional m_jointMotorTorque2 fields (to stay backward compatible, not breaking the existing C API) and filling them in PhysicsServerCommandProcessor.cpp
https://github.com/bulletphysics/bullet3/blob/master/examples/SharedMemory/PhysicsServerCommandProcessor.cpp

erwincoumans, 2019-02-27:
Fixed in this PR: https://github.com/bulletphysics/bullet3/pull/2130

closed 2019-02-27 by erwincoumans