bulletphysics/bullet3 #2125: Using pybullet GUI with matplotlib leads to NSInvalidArgumentException on macOS

jimmyyhwu, 2019-02-24:
I am using pybullet 2.4.2 with python 3.6 on macOS Mojave.

Code to reproduce:

```
import matplotlib.pyplot as plt
import pybullet as p

p.connect(p.GUI)
plt.imshow(100*[100*[0]])
plt.show()
```

Result:

```
python[3180:370161] -[NSApplication _setup:]: unrecognized selector sent to instance 0x7f997f83e0a0
python[3180:370161] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplication _setup:]: unrecognized selector sent to instance 0x7f997f83e0a0'
```

Some of the example files give the same error. For example, [getCameraImageTest.py](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/getCameraImageTest.py).

erwincoumans, 2019-02-25:
I just tried on Mojave with Python 3.7.1 (in /usr/local/bin/python3) and it works fine.
[image]

DId you enable the TkAgg?

import matplotlib
matplotlib.use("TkAgg")

or create a file called ~/.matplotlib/matplotlibrc
and add the text
~~~~~~
backend: TkAgg
~~~~~~

(I didn't need to do this)

jimmyyhwu, 2019-02-25:
Yes, I already have TkAgg in my `~/.matplotlib/matplotlibrc`. I just tried again with a new miniconda environment, installing python 3.7.2, matplotlib 3.0.2, pybullet 2.4.5, and running the below script.

```
import matplotlib
matplotlib.use("TkAgg")
import matplotlib.pyplot as plt
import pybullet as p

p.connect(p.GUI)
plt.imshow(100*[100*[0]])
plt.show()
```

Unfortunately, I still get the NSInvalidArgumentException.

erwincoumans, 2019-02-25:
Unfortunately this is likely a Matplotlib + Mojave issue, not something we can solve on the Bullet side. We just use simple OpenGL initialization. So I have to close it. Please report it in the Matplotlib support (if there is such a thing).

jimmyyhwu, 2019-02-25:
Thanks for the suggestions!

It looks like creating a matplotlib window using `plt.figure()` before connecting to GUI makes the error go away.

```
import matplotlib.pyplot as plt
import pybullet as p

plt.figure()
p.connect(p.GUI)
plt.imshow(100*[100*[0]])
plt.show()
```

With this change, [getCameraImageTest.py](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/getCameraImageTest.py) also works.

I can also confirm that the SHARED_MEMORY approach works too.

closed 2019-02-25 by jimmyyhwu