bulletphysics/bullet3 #2128: Missing functionality in C API

Harha, 2019-02-26:
Hey.

I've been using bullet3 for my private game engine project which is written in C. It's going well, but I've noticed the C API of bullet3 seems to lack many functionalities when compared to the full C++ 'API'. For example, I've had to create rigid bodies using the multi body functions, which is okay I guess but performance-wise slightly heavier than normal rigid bodies AFAIK.

Is there an ongoing work put into making this C API more complete? If no one is working on it at the moment, I could try to dedicate some of my free time into researching what are the main things missing from it and maybe come up with a branch that implements the missing functionalities.

AndresTraks, 2019-02-26:
Here's a C wrapper that is pretty much complete: https://github.com/AndresTraks/BulletSharpPInvoke/tree/master/libbulletc
However, it is mostly designed as an intermediate layer between C++ and other languages like C#, so I'm not sure how easy it is to use directly in C.

I've briefly discussed merging the two C APIs with Erwin, but haven't had time to follow up.

erwincoumans, 2019-02-26:
If you want to stay closer to the Bullet C++ internals, you may want to use @AndresTraks C API.
The BulletRobotics C API is abstracted to allow larger changes under the hood.

Note that you can still create btRigidBody with the BulletRobotics API. Just set 
~~~~~~~~
m_useMaximalCoordinates=true.
~~~~~~~~

closed 2019-02-26 by erwincoumans