bulletphysics/bullet3 #2135: [Question] on code snippet in robot_bases.py

junhyeokahn, 2019-03-01:
Hello.
What is the meaning of [this line](https://github.com/bulletphysics/bullet3/blob/c6cf5cb0c4bdda26466f7ad14f7812a8eb23c6c8/examples/pybullet/gym/pybullet_envs/robot_bases.py#L62).
Is it also disabling default motor for direct torque control as described in PyBullet Quick Start Guide?

benelot, 2019-03-02:
Yes. That is what it does.

junhyeokahn, 2019-03-02:
Thanks for the reply!
I have one more quick question.
If I comment out that line, there seems to be a lot of friction and the motor barely moves.
My understanding when I read the PyBullet Quick Start Guide was the command ```setJointMotorControl2``` has an effects on the single step calculation.
How does it work behind the scene?

erwincoumans, 2019-03-02:
In POSITION_CONTROL and VELOCITY_CONTROL and PD_CONTROL, the effect is permament.
Only in TORQUE_CONTROL, the torque is applied a single step. We probably should have removed the TORQUE_CONTROL and let the user manually apply a joint torque (which is the same). That would break the API and that's no good.

junhyeokahn, 2019-03-02:
That makes sense. Thanks!

closed 2019-03-02 by junhyeokahn