bulletphysics/bullet3 #2140: calculateInverseKinematics inaccurate tr

duyhungbkhn, 2019-03-04:


erwincoumans, 2019-03-04:
Please provide a small reproduction case.

duyhungbkhn, 2019-03-04:
When I use function "calculateInverseKinematics" to keep end effector follow circle trajectory in while loop with time.sleep(0.001), it seem inaccurate than I use bigger time.sleep(0.1). Is that for my computer configuration, is not strong enough to execute this function in high speed?

erwincoumans, 2019-03-04:
I tried reproducing this with the inverse_kinematics.py example, setting the timestep to 0.001, but it track just fine. Can you look at the attached example, and share your version that shows the issue?

[inverse_kinematics.zip](https://github.com/bulletphysics/bullet3/files/2927544/inverse_kinematics.zip)

duyhungbkhn, 2019-03-05:
I have used 'setRealTimeSimulation(1)'. This is my case I have problem:
[NadRobot.tar.gz](https://github.com/bulletphysics/bullet3/files/2928822/NadRobot.tar.gz)

erwincoumans, 2019-03-20:
Use setRealTimeSimulation(0) instead.

closed 2019-03-20 by erwincoumans