bulletphysics/bullet3 #2151: Create a bridge between ROS and PyBullet

erwincoumans, 2019-03-09:
See also https://pybullet.org/Bullet/phpBB3/viewtopic.php?f=24&t=12623

erwincoumans, 2019-03-09:
As alternative, there is Gibson, which uses PyBullet and its own rendering. It seems to be ROS bindings: https://github.com/StanfordVL/GibsonEnv

QingkaiLu, 2019-04-04:
It would be convenient for people who use ROS to have a bridge between ROS and PyBullet.

Pandhariix, 2019-04-07:
I agree, it would be very convenient. In our project [qibullet](https://github.com/ProtolabSBRE/qibullet) we ended up creating a wrapper class, creating a bridge between ROS and our simulated model

oscar-lima, 2019-06-16:
started a pybullet ROS wrapper [here](https://github.com/oscar-lima/pybullet_ros), still in early stage but help and coments are welcome ; )

erwincoumans, 2019-06-18:
Thanks @oscar-lima and self-contained pybullet ROS wrapper would be nice. What features do you plan to wrap? I imagine there could be control (setJointMotorControl), sensors (joint angle encoder, camera vision (getCameraImage), LIDAR (using batch ray cast), touch/contact (using getContactPoints).

erwincoumans, 2019-06-18:
Also related to ROS+PyBullet is this project:
https://github.com/ProtolabSBRE/qibullet

oscar-lima, 2019-06-18:
> What features do you plan to wrap?

Control and sensors was on my list, but I welcome your suggestions. Thanks!

oscar-lima, 2019-12-11:
> I have started a pybullet ROS wrapper [here](https://github.com/oscar-lima/pybullet_ros), still in early stage but help and comments are welcome ; )

@erwincoumans basic implementation is done, more urgently what I am missing is the parallelization of the plugin in threads. Can you please take a look and provide some feedback?

erwincoumans, 2019-12-11:
What do mean by parallelization? Is this a C plugin for PyBullet or a Python plugin for ROS?
Usually only one thread/process should communicate with PyBullet at one time, so you need to use some synchronization. There are two shared memory channels open by default, so you can use the second from a different thread as well.

oscar-lima, 2019-12-13:
> Thanks for the update, that looks great from a first glance!

Thanks! :D

> 
> What do mean by parallelization? Is this a C plugin for PyBullet or a Python plugin for ROS?

pybullet plugins? Is something that I invented, pseudocode goes like this:

```
while application_alive:
    --- apply plugins in sequence (no threading yet) ---
    e.g. read odometry and publish to ROS (odometry  plugin)
    e.g. read user joint command from ROS and apply setJointMotorControlArray(pos, vel or torque) to robot (control plugin)
    e.g. use rayTestBatch to query laser hits on pybullet and publish ROS laser msg (laser plugin)
    -------------------------------------------------------------------
    step simulation # (self.pb.stepSimulation())
```

Note all pybullet plugins share the same import, for that I use importlib like this:

```
self.pb = importlib.import_module('pybullet')
```

and then I share self.pb to dynamically loaded plugins (also using importlib iterating over the required plugins)

> Usually only one thread/process should communicate with PyBullet at one time, so you need to use some synchronization. There are two shared memory channels open by default, so you can use the second from a different thread as well.

Here is where I have the main problem, because e.g. the laser task (rayTestBatch) takes a long to compute and slows down the flow. The synchronization you mention is something I would have to implement manually right? Can you provide with an example?

closed 2021-04-26 by erwincoumans