bulletphysics/bullet3 #2152: Simulating fast time-scale dynamics

avikde, 2019-03-09:
I have created this simple example in pybullet together with a simple 1DOF urdf: https://gist.github.com/avikde/edc5b54cb54e1a7e5bc4c89c76c32dd2

This is the resulting desired(blue) and actual (orange) position:

[image]

I (think--you can see the pybullet usage) I have allowed it unlimited force for this tracking, alleviated URDF joint limits, but in my actual example, as well as this one, the resulting positions have these regions of what appears to be constant velocity ("linear" regions in the position plot), in addition to the reduced amplitude.

I have tried
- with and without specifying target velocity
- with different position/vel gains

Do you know what I could try to enable better tracking for this kind of example?

ThorMalleuson, 2019-03-09:
Your simulation is likely CPU bound, or you have a set time step that is too slow for your simulation.  What you are seeing is called aliasing. https://en.wikipedia.org/wiki/Aliasing

erwincoumans, 2019-03-09:
There are a few issues:

1) You are using PD gains, but those are not suitable for POSITION_CONTROL mode, but you can use PD gains for the PD_CONTROL control method. For POSITION_CONTROL use Kp=1 and Kd=0.1, gains need to be in range [0..1].

2) In addition, your velocity exceed 100 rad/sec, and this hits a safety clamp. I just exposed a method to increase this maxJointVelocity setting.
~~~~~~~~~~~~~~~~
p.changeDynamics(bid,	-1,	maxJointVelocity=1000)
~~~~~~~~~~~~~~~~

~~~~~~~~~~~~~~~~~~~
	usePD	=	False
	if usePD:
		p.setJointMotorControlArray(bid, [1],	p.PD_CONTROL,	targetPositions=[qdes],	targetVelocities=[dqdes],	positionGains=[100000],	velocityGains=[100], forces=[1000000000])
	else:
		p.setJointMotorControlArray(bid, [1],	p.POSITION_CONTROL,	targetPositions=[qdes],	targetVelocities=[dqdes],	positionGains=[1], velocityGains=[0.1],	forces=[1000000000])
~~~~~~~~~~~~~~~~~~~

[image]

Just in case, even though this doesn't affect your test noticeably, but you can disable default damping like this:
~~~~~~~~~~~~~~~~~~~~~
p.changeDynamics(bid,-1,linearDamping=0, angularDamping=0)
~~~~~~~~~~~~~~~~~~~~~

See attached avik.zip for a fix. You will need a new PyBullet for the changeDynamics(maxJointVelocity).
https://github.com/bulletphysics/bullet3/pull/2153

[avik.zip](https://github.com/bulletphysics/bullet3/files/2948645/avik.zip)

avikde, 2019-03-10:
Thanks @erwincoumans that fixes my problems!

closed 2019-03-10 by avikde