bulletphysics/bullet3 #2157: Incorrect contact points

mburetorp, 2019-03-11:
While trying to make a cylinder roll straight on a triangulated plane, I noticed that contact points sometimes are generated far away from the actual cylinder. These contacts are still processed so the distance calculation is not correct.

I've attached a modified Hinge2Vechile demo to illustrate this. Contact points are visualized as pink lines in gContactAddedCallback and blue lines in gContactProcessedCallback.

This prevents the cylinder from rolling smoothly, even when forcing the contact normal to (0, 1, 0) in the callback. Any idea what might cause this and how to fix it? I can debug this further with some help (I'm not familiar with GJK).

[contactbug.zip](https://github.com/bulletphysics/bullet3/files/2952361/contactbug.zip)

[image]

mburetorp, 2019-03-11:
Sorry, I forgot add the mesh file in the zip so the demo will not work. I'll fix that tomorrow.

mburetorp, 2019-03-12:
Please see the attached demo: [contactbug.zip](https://github.com/bulletphysics/bullet3/files/2955666/contactbug.zip)

erwincoumans, 2019-03-12:
I had a quick look. The issue is due to floating point cancellation inside GJK. If you compile the code using double precision, it works fine. You can use premake to build it:

cd build3
./premake4(_linux/osx) --double vs2010/gmake or use cmake to enable double precision.
~~~~~~~~~~
btIndexedMesh part;
part.m_vertexType=PHY_FLOAT;
~~~~~~~~~~

One solution would be to compile the critical parts of GJK always in double precision mode (not using btScalar but using double, inside GJK).

mburetorp, 2019-03-14:
I recompiled using double precision and yes it works fine, which is good! There was a 20% performance hit however (on Ryzen Threadripper), not from doubles I think but from not using any SSE code. I got a similar performance hit with single precision and no SSE.

Theoretically I think that GJK with any meter sized objects should work reliable without the use of double precision, but then again I have never worked with GJK before.

> One solution would be to compile the critical parts of GJK always in double precision mode (not using btScalar but using double, inside GJK).

I can try that. Where in the code base is this critical part?

erwincoumans, 2019-03-20:
Check out btGjkEpa2.cpp, btGjkPairDetector.cpp and btVoronoiSimplexSolver.cpp\
If you have a patch, consider opening a pull request.

closed 2019-03-20 by erwincoumans