bulletphysics/bullet3 #2158: maybe the return objects are not expected them in line 289 of kuka_diverse_object_gym_env.py

TomoyaFukui, 2019-03-12:
Thank you for the excellent simulator.
It is really helpful and interesting for me to implement some reinforcement algorithms.
By the way, related to the lines from 316 to 320, glob.glob returns the same result if the test is True or False.
I think that if revising line 319 to "urdf_pattern = os.path.join(self._urdfRoot, 'random_urdfs/*[1-9]/*.urdf')", it returns your expected results.
I hope this helps you.
'''
316 if test:
317       urdf_pattern = os.path.join(self._urdfRoot, 'random_urdfs/*0/*.urdf')
318     else:
319       urdf_pattern = os.path.join(self._urdfRoot, 'random_urdfs/*[^0]/*.urdf')
320 found_object_directories = glob.glob(urdf_pattern)
'''

erwincoumans, 2019-03-12:
Thanks for the report, did this behavior change in Python 3.6.8?
Yes, we can make this change. Also, I'd like to load those files directly from a single zip file, but that requires a few more lines of change (those thousands of random urdfs in the make install/setup slow). PyBullet can load files directly from a zip archive.

TomoyaFukui, 2019-03-13:
This is the problem of glob module. Apparently, glob function ignores a caret in the regular expression.
That's why glob.glob returns the same results if the test is true or not.
I don't know how to confirm glob version, but this code can check the results that we mention. 
I'm sorry if this code works well.
```
import glob
import pybullet_data

path = pybullet_data.getDataPath()
urdf_pattern = path + "/random_urdfs/*[0]/*.urdf"
urdf_pattern_caret = path + "/random_urdfs/*[^0]/*.urdf"
urdf_pattern_1to9 = path + "/random_urdfs/*[1-9]/*.urdf"
print(urdf_pattern)
print(urdf_pattern_caret)
print(glob.glob(urdf_pattern)==glob.glob(urdf_pattern_caret)) # True
print(glob.glob(urdf_pattern)==glob.glob(urdf_pattern_1to9)) # False
```

erwincoumans, 2019-03-14:
Great, thanks for tracking this down. Here is a pull request with a reference to you/this issue:
https://github.com/bulletphysics/bullet3/pull/2159

closed 2019-03-15 by TomoyaFukui