bulletphysics/bullet3 #2163: delete saved states in PyBullet

zeshiYang, 2019-03-19:
Hi,
I use p.saveStates() function to save some states in simulation and I want to delete some of them, however I didn't find any functions can do that, it seems that resetsimulation() will delete all the states saved in the physics engine... How should I cope with that?(I don't want to save states in Disk...) Thank you!

erwincoumans, 2019-03-20:
Sure, we can add a removeState. If it is urgent for you and you are familiar with C++, you could try adding it yourself. It will touch a few files, pybullet.c for python binding, PhysicsServerCommandProcessor.cpp for the remove and the BulletClientC_Api.h/cpp.

erwincoumans, 2019-04-04:
Implemented here: https://github.com/bulletphysics/bullet3/pull/2182

closed 2019-04-04 by erwincoumans