bulletphysics/bullet3 #2164: Memory leak when using inverse kinematics

amandlek, 2019-03-19:
I'm trying to use inverse kinematics for the Sawyer arm, but I'm running into issues with memory leaks. [Here](https://www.dropbox.com/s/oqazgx7v20h1ps6/bullet_test_ik.zip?dl=0) is a zip file that contains a minimal python script and a set of URDFs to reproduce the issue. You should just be able to unzip the folder, change directory, and run `python test_bullet.py`. To monitor memory usage, open another terminal and run `watch -d free -m`. 

The code that I've been using was originally based on version `1.9.5` where running IK in a loop was beneficial but I've been able to reproduce the memory leak on the latest version as well, along with a single iteration of IK. 

Could someone please advise? The memory leak is preventing us from using pybullet IK as part of an environment for training an RL agent.

erwincoumans, 2019-03-20:
Thanks for the repro case, I'll look into it soon (give it a week or so) and if reproducable fix it.

erwincoumans, 2019-03-20:
It is fixed now, after it passes continuous integration tests, it will propagate to pypi soon.
https://github.com/erwincoumans/bullet3/commit/f725d1201d375bce529448abd9dec1f00005026e

closed 2019-03-21 by erwincoumans

amandlek, 2019-03-21:
Thanks for fixing the issue so quickly!