bulletphysics/bullet3 #2166: static object world transform not updated in its motion state...

MilaCridlig, 2019-03-20:
For rendering, I happened to be using **m_motionState->getWorldTransform(transform)**,
rather than **m_rigidBody->getWorldTransform()**.

By definition, the motion state for a static object is not updated, so the copy of the world transform in its motion state was apparently not changed when the world transform of the corresponding rigid body was changed, while it should have been for consistency.

[image]

erwincoumans, 2019-03-20:
Indeed, the sync transforms doesn't apply to static objects, best to only use them for dynamic objects.

closed 2019-03-20 by erwincoumans

MilaCridlig, 2019-03-22:
Still, for consistency, it would be good to update the world transform of the motion state even for static objects in case it is queried for rendering instead of the world transform of the rigid body which is the only one updated as of now for static objects.