bulletphysics/bullet3 #2170: [pybullet] How to disable b3Warning

iory, 2019-03-23:
Thanks for great works!
By the way, I would like to know how to disable b3Warning message in pybullet.
I would like to know if there is the method.

erwincoumans, 2019-03-26:
There is no such feature. Best to try dealing with the warning so it doesn't show up.

closed 2019-03-26 by erwincoumans

iory, 2019-03-26:
OK. Thanks!

ethanabrooks, 2021-11-19:
I know that this is long closed but first, let me say that there are reasons why it may be prohibitive to deal with warnings. In my case, I am iteratively loading from a dataset of more than 80 URDF files  (not created by me) and which seem to never provide inertial data for the links, resulting in warnings like the following:

```
link_1b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
```

 In my setting, the models corresponding to the URDF files need not respond to physical forces so it is a non-issue to me. Meanwhile, I am not going to go through and correct 80-odd URDF files by hand just to suppress a warning.

So for those who find themselves in a similar position, the following (borrowed from https://stackoverflow.com/a/17954769/4176597) worked for me:

```python
@contextmanager
def suppress_stdout():
    fd = sys.stdout.fileno()

    def _redirect_stdout(to):
        sys.stdout.close()  # + implicit flush()
        os.dup2(to.fileno(), fd)  # fd writes to 'to' file
        sys.stdout = os.fdopen(fd, "w")  # Python writes to fd

    with os.fdopen(os.dup(fd), "w") as old_stdout:
        with open(os.devnull, "w") as file:
            _redirect_stdout(to=file)
        try:
            yield  # allow code to be run with the redirected stdout
        finally:
            _redirect_stdout(to=old_stdout)  # restore stdout.
            # buffering and flags such as
            # CLOEXEC may be different
```
and then 
```python
  with suppress_stdout():
      model = self._p.loadURDF(path).  # This line used to cause the warning.
```

albuckley-leap, 2025-02-20:
> with suppress_stdout():

I tired this but all it does is temporarily supresses the output and it all bounces back as soon as I close the simulation.