bulletphysics/bullet3 #2173: [pybullet] Gravity acts along wrong axis in VR - but only in VR.

sholtodouglas, 2019-03-24:
Hi Erwann,

When I run the same set of code through the normal GUI - gravity acts correctly as specified in the z axis. However, when I launch a shared physics server and connect to this - gravity acts along a different axis. 

I think the axis are still the same as the handheld controller behaves correctly - when I read out it's xyz coordinates moving it up and down affects the z coordinate. I've intentionally made gravity lower because otherwise the jenga piece races away and it's hard to find it in VR before it's gone. 

Any ideas on how I should begin finding the cause of this?

Best, Sholto
```

def basic_vr_env():
    cid = p.connect(p.SHARED_MEMORY)
    print("Connected to shared memory")
    if (cid<0):
        p.connect(p.GUI)
    p.resetSimulation()
    p.setGravity(0,0,-0.10)
    #disable rendering during loading makes it much faster
    p.configureDebugVisualizer(p.COV_ENABLE_RENDERING, 0)
    p.setVRCameraState([0.5,0,0],p.getQuaternionFromEuler([0,0,-90]))
    objects = [p.loadURDF((os.path.join(urdfRoot, "plane.urdf")), 
     0.000000,0.000000,0.000000,0.000000,0.000000,0.000000,1.000000)]
    objects = [p.loadURDF((os.path.join(urdfRoot, "jenga/jenga.urdf")), 
    1.300000,-0.700000,0.750000,0.000000,0.707107,0.000000,0.707107)]
    objects = [p.loadURDF((os.path.join(urdfRoot, "table/table.urdf")), 1.000000,-0.200000,0.000000,0.000000,0.000000,0.707107,0.707107)]
    p.configureDebugVisualizer(p.COV_ENABLE_RENDERING, 1)
    p.setRealTimeSimulation(1)
    while (1):
        print('-')
        continue

    p.disconnect()`
```

erwincoumans, 2019-03-26:
There is a command to configure the visualizer to change the UP axis.
p.configureDebugVisualizer(p.COV_ENABLE_Y_AXIS_UP ,1/0). Have you tried this?

closed 2019-03-26 by erwincoumans

sholtodouglas, 2019-03-30:
I just tried both of those options, but they only modify the orientation /position of my headset's viewpoint. The issue seems to be that the axis are rotated, in the normal sim the blue axis  (z axis) is pointing upward, out of the plane, but in VR it is the green axis. Is there anywhere these could have been rotated?

sholtodouglas, 2019-04-07:
I ended up configuring my normal sim with the Y axis up - gravity still acts in the wrong direction in VR (the z axis must be set for it to act along the Y axis), but at least it lines up with the sim now.  I had to transform the controller position and orientation to Y axis up, but that wasn't so bad - thanks for the help!