bulletphysics/bullet3 #2175: Cannot find .dae file of ROS urdf model

arwtyxouymz, 2019-03-26:
Hi, I'm trying to spawn ur5 robot model to pybullet.
I use [universal robot ROS package](https://github.com/ros-industrial/universal_robot) and included this model to my robot model.

Because this package's urdf is written in `xacro`, firstly I converted to urdf, then it loaded by `loadURDF`.
But I encountered these error.

```
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/UrdfFindMeshFile.h,108]:

b3Printf: /home/ryutaro/work/catkin_ws/src/ur5_robotiq85/ur5_robotiq85_description/robot/ur5.urdf:44: cannot find 'ur_description/meshes/ur5/visual/base.dae' in any directory in urdf path

Warning: b3Error[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,121]:

Warning: Could not parse visual element for Link:
Warning: b3Error[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,121]:

Warning: base_link
Warning: b3Error[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,121]:

Warning: failed to parse link
Traceback (most recent call last):
  File "/home/ryutaro/work/catkin_ws/src/ur5_robotiq85/ur5_robotiq85_pybullet/src/main.py", line 30, in <module>
    boxId = p.loadURDF(urdf_path, cubeStartPos, cubeStartOrientation)
pybullet.error: Cannot load URDF file.
```

It looks urdf parser cannot find `.dae` file.
In converted urdf file,
```xml
      <geometry>
        <mesh filename="package://ur_description/meshes/ur5/visual/base.dae"/>
      </geometry>
```
`package://` description are used for rospack? can find this `.dae` file from everywhere.

How solve this?
Is it impossible to use ROS URDF model just the way it is in pybullet?

erwincoumans, 2019-03-26:
There is no URDF in https://github.com/ros-industrial/universal_robot
What sub folder?
COLLADA/DAE is a complicated format that isn't well supported. It is best to convert it to Wavefront OBJ.

arwtyxouymz, 2019-03-26:
@erwincoumans 

I'm sorry, this package [ur_description](https://github.com/ros-industrial/universal_robot/tree/kinetic-devel/ur_description/urdf).

Is there any practice to convert to Wavefront OBJ?

erwincoumans, 2019-03-26:
You can use Blender to import the DAE and export to OBJ, or use meshlab.
Good luck, we cannot provide support with all kinds of URDF file issues.

closed 2019-03-26 by erwincoumans