bulletphysics/bullet3 #2185: Question: Maximal number of objects in VisualShapeArray

vctrs, 2019-04-04:
Hello

Is there a limit on the number of bodies which can used to create a visual shape array? I have a list of 18 tetrahedrons saved as obj files which have to be used to create a single body. 

CreateVisualShapeArray can take random 16 of them (meaning the problem is not with files themselves). If I try to use more it throws an exception "createVisualShapeArray failed".

CreateCollisionShapeArray takes 18, but seems to only use 16 of them.

Does this limit on the number of used meshes exist? If so, what would be the workaround?

Thanks

erwincoumans, 2019-04-04:
Yes, the maximum number of compound/visual shapes in 16.
You can modify this number in the source code here: https://github.com/bulletphysics/bullet3/blob/31ce4e0e57ddb2bc8a23ebc371b627eaff813aa3/examples/SharedMemory/SharedMemoryCommands.h#L952

vctrs, 2019-04-04:
Thanks a lot!

vctrs, 2019-04-04:
I changed it in SharedMemoryCommands.h. But it didn't solve the issue...

erwincoumans, 2019-04-04:
It should, how did you install pybullet with those changes?

vctrs, 2019-04-05:
Properly reinstalled and now it seems to work! Thanks

closed 2021-04-26 by erwincoumans